motioneye-project/motioneyeos #2: No email notifications after installing motioneyeOS

SuperSharpShooter, 2015-10-13:
I've just installed the new version, and motion detection and image and video capture are working fine, however no emails are being sent.

I've got it set to use gmail to send the images, and it is set up exactly as it was with the previous version, but it seems there's no attempt for the emails to be sent. I can't see anything in the logs that shows any attempt, or any issues with sending the emails.

Link to logs: https://www.dropbox.com/s/y33mtwfpj8ibu38/motioneye.zip?dl=0

mikecarr, 2015-10-16:
I am having the same exact issue and I see nothing in the logs suggesting it is even trying to send email.

arxat, 2015-10-20:
I also receive notifications by email, I have everything configured properly and that motionpie if it works.

arxat, 2015-10-20:
After doing some tests I have seen with notifications if I Raspicam but not an IP camera through RTSP

SuperSharpShooter, 2015-10-20:
I'm using the Raspicam, but am getting no email notifications.

ccrisan, 2015-10-20:
The issue has been fixed upstream in motionEye with version 0.28.2. It will soon be included in the next motionEyeOS version.

closed 2015-10-20 by ccrisan