motioneye-project/motioneyeos #3: Email notification problem

Feuxxxeu, 2015-10-14:
Hello,
I am behind a router, so the raspberry pi B has access to internet.
Please advise
I've been trying with gmail account: 
2015-10-13 20:49:17: [sendmail]    ERROR: failed to send mail: (534, '5.7.14 <https://accounts.google.com/ContinueSignIn?sarp=1&scc=1&plt=AKgnsbsfE\n5.7.14 dovOr0AipLQF0Pp9SxNPSxSnf3QRcCAIozB--blrclJaBDd50LWA-gBl4ALCJHj4T8i35O\n5.7.14 I-Xa7llM3OhcYelEQCZYSJAKHDlQ5bIxDlOq5EflSY6EM1LcSO8iiuiIPveZWeOIyv28mf\n5.7.14 VPEvAXdLC6oRtzybdyzjdeSsy8bGQnJ45SsKYgR1N9sKcMX3_YPhHAuZBXELPY6xwbmX6i\n5.7.14 5ypMmdBHrV0jlq_eQtsobBtzeEjY> Please log in via your web browser and\n5.7.14 then try again.\n5.7.14  Learn more at\n5.7.14  https://support.google.com/mail/answer/78754 gt4sm16082779wib.21 - gsmtp')
Traceback (most recent call last):
  File "/home/ccrisan/motioneyeos/output/raspberrypi/target/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 178, in on_message
  File "/home/ccrisan/motioneyeos/output/raspberrypi/target/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 54, in send_mail
  File "/usr/lib/python2.7/smtplib.py", line 622, in login
SMTPAuthenticationError: (534, '5.7.14 <https://accounts.google.com/ContinueSignIn?sarp=1&scc=1&plt=AKgnsbsfE\n5.7.14 dovOr0AipLQF0Pp9SxNPSxSnf3QRcCAIozB--blrclJaBDd50LWA-gBl4ALCJHj4T8i35O\n5.7.14 I-Xa7llM3OhcYelEQCZYSJAKHDlQ5bIxDlOq5EflSY6EM1LcSO8iiuiIPveZWeOIyv28mf\n5.7.14 VPEvAXdLC6oRtzybdyzjdeSsy8bGQnJ45SsKYgR1N9sKcMX3_YPhHAuZBXELPY6xwbmX6i\n5.7.14 5ypMmdBHrV0jlq_eQtsobBtzeEjY> Please log in via your web browser and\n5.7.14 then try again.\n5.7.14  Learn more at\n5.7.14  https://support.google.com/mail/answer/78754 gt4sm16082779wib.21 - gsmtp')
2015-10-13 20:50:06: [sendmail]    ERROR: failed to send mail: (534, '5.7.14 <https://accounts.google.com/ContinueSignIn?sarp=1&scc=1&plt=AKgnsbv-0\n5.7.14 VUSJavXzBie_XgnuiogePombkrUWjLQ-dDhfhbiSCaPbFFESUkAXdNo0EwigP84wlWwdDx\n5.7.14 NStXJViC_qHv_XL3zLU-uvVsXIrfpstVbs0pv9nUGwYEbgZoqSzqSUv9_n7Jk41R5GCPwo\n5.7.14 s1KWbHIOswJAaZQO9G1kQ7ez_E1h9LIeSzUtjZ6YqWONt7eaLNPPdG4x4XDnyBDPwyEiiy\n5.7.14 svNKCff8fm35Gfr5_q8_QLcULTtg> Please log in via your web browser and\n5.7.14 then try again.\n5.7.14  Learn more at\n5.7.14  https://support.google.com/mail/answer/78754 ub7sm3945435wib.17 - gsmtp')
Traceback (most recent call last):
  File "/home/ccrisan/motioneyeos/output/raspberrypi/target/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 178, in on_message
  File "/home/ccrisan/motioneyeos/output/raspberrypi/target/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 54, in send_mail
  File "/usr/lib/python2.7/smtplib.py", line 622, in login
SMTPAuthenticationError: (534, '5.7.14 <https://accounts.google.com/ContinueSignIn?sarp=1&scc=1&plt=AKgnsbv-0\n5.7.14 VUSJavXzBie_XgnuiogePombkrUWjLQ-dDhfhbiSCaPbFFESUkAXdNo0EwigP84wlWwdDx\n5.7.14 NStXJViC_qHv_XL3zLU-uvVsXIrfpstVbs0pv9nUGwYEbgZoqSzqSUv9_n7Jk41R5GCPwo\n5.7.14 s1KWbHIOswJAaZQO9G1kQ7ez_E1h9LIeSzUtjZ6YqWONt7eaLNPPdG4x4XDnyBDPwyEiiy\n5.7.14 svNKCff8fm35Gfr5_q8_QLcULTtg> Please log in via your web browser and\n5.7.14 then try again.\n5.7.14  Learn more at\n5.7.14  https://support.google.com/mail/answer/78754 ub7sm3945435wib.17 - gsmtp')
2015-10-13 20:51:43: [sendmail]    ERROR: failed to send mail: [Errno 97] Address family not supported by protocol

And Yahoo Account:
2015-10-14 07:10:45: [sendmail]    ERROR: failed to send mail: Connection unexpectedly closed: timed out
Traceback (most recent call last):
  File "/home/ccrisan/motioneyeos/output/raspberrypi/target/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 178, in on_message
  File "/home/ccrisan/motioneyeos/output/raspberrypi/target/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 49, in send_mail
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
  File "/usr/lib/python2.7/smtplib.py", line 317, in connect
  File "/usr/lib/python2.7/smtplib.py", line 365, in getreply
SMTPServerDisconnected: Connection unexpectedly closed: timed out

Also with my ISP account, all with or without TLS, tried ports 25, 465 and 587.

ccrisan, 2015-10-14:
I found this online: http://stackoverflow.com/questions/20337040/gmail-smtp-debug-error-please-log-in-via-your-web-browser.

Feuxxxeu, 2015-10-14:
Thank you very much !
2015-10-14 19:50:38: [sendmail]     INFO: email sent
It works with gmail now.
I had to go to https://www.google.com/settings/security/lesssecureapps to allow less secure apps to be used.
Then https://accounts.google.com/DisplayUnlockCaptcha to unlock the next try of a non-secure app smtp send
This is a little disturbing to leave this less secure apps option on, i'll continue fiddling with other stmp relays.

closed 2015-10-22 by ccrisan

psts, 2015-10-24:
I have 2-step verification turned on, on my Gmail account. Then you can issue specific app passwords that you can use for things like this. Works like a charm and highly secure.