motioneye-project/motioneyeos #4: How to synchronize with Google Drive

malcolmct, 2015-10-15:
motionEyeOS is great, but having seen a number of requests for Google Drive functionality, I thought I should share how I got it working.  

Here are step-by-step instructions which also work with motionPie:

-	Ensure that you have set a motionEyeOS admin password through the web interface and have rebooted

-	Using SSH, login as root with your admin password from a Mac, Linux or Windows PC

-	Enter and execute the following commands.  You can do so one line at a time or just copy and paste
	the whole lot into your SSH session:
```bash
# mount the root volume as read/write
mount -o remount,rw /
# change to the /usr/bin directory where we will keep the drive executable
cd /usr/bin
# download a pre-built gdrive binary file
wget https://github.com/odeke-em/drive/releases/download/v0.2.2-arm-binary/drive-arm-binary
# make it executable
chmod +x drive-arm-binary
# rename it to gdrive
mv drive-arm-binary gdrive
# change the root volume back to read-only
mount -o remount,rw /
# create a gdrive directory on the data volume
mkdir /data/gdrive
# change to that directory
cd /data/gdrive
# download my modified version of Claude Pageau's sync.sh script from Dropbox
wget https://www.dropbox.com/s/uiipkef0gn3lm7x/sync.sh
# make the script executable
chmod +x sync.sh
# create a symbolic link to the motion output folder
ln -s /data/output output
```

-  Now you'll want to link gdrive to your Google account.   Open a browser on your PC and ensure
    that you are logged into Google (for example, you could open Gmail).
    
    Back in your SSH session enter and execute:
    
    `gdrive init`

      
    This will produce a long, complex URL which you should copy and paste into your PC browser's 
    address line, then press Enter.
      
    You'll need to give permission to manage files in your Google Drive and will then be returned a 
    code to copy back into your SSH session where gdrive is waiting.
    
    A success will return nothing.
    
-   Finally, you should add a cron entry to copy over files from the motion output folder to Google
	Drive on a regular schedule.  

```bash	  
	# set the crontab editor to be nano (I'm much more comfortable with nano than vi)
	  export EDITOR=nano
	# enter crontab EDITOR
	  crontab -e
```
-  Now set the task to run every five minutes by pasting in:
	
    `*/5 * * * * /data/gdrive/sync.sh >/dev/nul`
		
    Then quit nano with control-x and save the crontab entry.
	
That's it.  You can have it up and running in about five minutes and it seems to work without any
unforseen problems.

ccrisan, 2015-10-15:
Nice. Thanks!

Volyova, 2015-10-16:
Looks awesome. Will try that out.

AlRiley, 2015-10-21:
Followed the instructions to the letter and it worked a treat.  Many thanks

AlRiley, 2015-10-23:
Interesting, for some reason the process to push the jpeg to Google drive dies after 24 hours.  I had "Preserve Pictures" in Still Images set to one day.  Any ideas as to why?

Limeliz, 2015-11-08:
Awesome! I just went through all steps and althought I've had a RPI for just two days and never used any Linux-system before everything seems to be working. gdrive is uploading all photos and video right now. :-) :+1:

russellgeoff, 2015-11-12:
Thanks for these instructions! I really like being able to reference and store my photos on my Google Drive. 

I am having an issue when the images are uploaded to Google Drive. It appears that each upload generates a new folder for the images. This has caused my output folder to explode with repeated folders. It also appears that the same images are being uploaded to multiple folders repeatedly. I've included images of the issues below. Any help resolving this would be great! Thank you!

Folders on my PC:
[image]

Folders on Google Drive:
[image]

Same image uploaded multiple times:
[image]

Limeliz, 2015-11-19:
I've ran into two problems. Everything worked for a couple of days, but then it stopped working.

1. The photos has stopped uploading to Google Drive. Last activity was 12th november. When runnig sync.sh manually I see that gdrive is unable to connect to the internet. Althought I'm connected to the same wifi as the RPI and I'm able to connect.

2. When I ran sync.sh manually I saw an error saying that 'pgrep' could not be found (line 17 in the script). When trying to run pgrep manually I see that indeed it's not on the RPI. I'm new to *nix systems, but Googling has taught me what it's used for. Is there a way to download this tool from someplace so I can use it in MotoinEyeOS?

MotionUK, 2015-11-19:
Good Morning Limeiz,

Looking at the cron task i noticed that the data dump does not point to the correct location.

"*/5 * * * * /data/gdrive/sync.sh >/dev/nul"

the /dev/nul should read as/dev/null.....

also look at 

http://unix.stackexchange.com/questions/163352/what-does-dev-null-21-mean-in-this-article-of-crontab-basics

I'm currently sitting a Linux When I looked at the device folder (/dev) the /nul was not present in the OS.

Worth a shot............

MotionUK

MotionUK, 2015-11-19:
@malcolmct - thanks for an amazing contribution.   MotionUK

Limeliz, 2015-11-19:
@MotionUK Yeah, I noticed that "/dev/nul" was typed wrong, after googling about it for a few minutes. I've already corrected that. The weird thing is that it worked for a couple of days and then suddenly stopped working. I'll try and reinstall the whole thing and see if that helps. I'll get back here with the results. There shouldn't exist any folder called /dev/null, it's a "black hole" which is commonly used on *nix systems to dump data which you don't want to use. It simply makes sure that any output sync.sh produces is being discarded and not displayed.

Limeliz, 2015-11-20:
It seems like my cron wasn't running. The following made gdrive start syncing.
    crond stop
    crond start

I also edited the sync.sh script a bit, because I noticed it ran gdrive three times every cron instance. By using this (http://unix.stackexchange.com/questions/22044/correct-locking-in-shell-scripts) method with a lockfile I make sure the script is only running one instance.

@russellgeoff I also had a problem with duplicate files and folders. I'm not sure if it was because gdrive was running multiple instances, but adding `-ignore-name-clashes` to the gdrive-line in `sync.sh` removed all duplicates.

malcolmct, 2015-11-20:
@Limeliz Many thanks for your comments and improvements!

I've modified my original posting to incorporate Limeliz's changes.  I also point to his version of the sync.sh script.  I think I must have originally tested mine on Raspian and hadn't noticed the absence of pgrep in MotionEyeOS.

mastix, 2016-01-17:
I've updated the Limeliz's sync script to also pull changes (including deletions) from Google Drive. This allows to delete files and folders in Google Drive and these files will then be deleted from MotionEyeOS as well.

Just in case you're interested: https://gist.github.com/mastix/1a93fb8c6114a3ba79ed

parth15041995, 2016-07-13:
amazing work
thanks for sharing
In raspberry pi 2 works fine everything is perfect.

xhain, 2016-12-19:
Thanks for this!

bortek, 2017-02-23:
I have added a separate page on Wiki with the How-To above therefore closing this issue.

closed 2017-02-23 by bortek

Johnrasp, 2019-07-26:
There is no 'wget' in motionEYE OS
ref google drive #4
wget https://github.com/odeke-em/drive/releases/download/v0.2.2-arm-binary/drive-arm-binary
error is -sh: wget: command not found

jasaw, 2019-07-26:
Use curl.
curl -o <url>