motioneye-project/motioneyeos #5: Help installing alsa-utils

GizmoXomziG, 2015-10-16:
Hey Calin, motioneyeos really became great over the years. I'm currently running it on Raspberry Pi 2, and it's working great.

I am currently trying to add support for playing sounds (a request was made on https://github.com/ccrisan/motionpie/issues/180).
snd-usb-audio module is loaded in the kernel.
aplay is not installed by default, it seems the whole alsa-utils package isn't installed.
I did find motioneyeos/package/alsa-utils/Config.in & alsa-utils.hash & alsa-utils.mk in the master branch, so I'm guessing there is some way of enabling alsa-utils. What am I missing here?
Motioneyeos lacks apt-get or any similar tool, so I currently have no way of installing alsa-utils.

Please help me install alsa-utils. If there's an easy solution, please add instructions in your wiki/Tweaks.

Thank you :)

ccrisan, 2015-10-16:
You cannot install anything on motionEyeOS. It's not a general purpose OS; you can think of it rather as a firmware. Sound support is coming but is not yet present. The drivers have already been included in the kernel but the utility packages and anything related to sound is still absent.

If you want to play a sound or do something that's not currently available with motionEyeOS, please use a general purpose distro (such as Raspbian).

GizmoXomziG, 2015-10-16:
Oh, now I get it - motionEye is the frontend for motion, and motionEyeOS is a minimal distro for motionEye.

Any ETA on sound support? Just curious, no pressure :)

Thanks for the fast response. Heading to https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian now :)

ccrisan, 2015-10-16:
No ETA and just so you know, this is definitely not a priority. Check out the roadmap for an idea about the priorities: https://github.com/ccrisan/motioneyeos/wiki/Roadmap.

Jonesinho, 2015-10-27:
Is an audio tweak planned or does other similar repos with audio support?

closed 2017-02-23 by bortek