motioneye-project/motioneyeos #9: Crashing .. solid green led

scoobyd00, 2015-10-21:
My system keeps crashung and hangs, unable to ssh or access webpage. I get a solid green light and have to power cycle the pi.
Its not consistant, it can run for a day or so then crash and other times for only a few hours.
Ive got images saving to a network share so the sd card shouldnt be the issue although i thought solid green led meant sd boot issue?.
Cant seem to see anything in  the logs
Any ideas?

scoobyd00, 2015-10-21:
Reset by a power cycle and its hung again after only 2hrs this time.. I've got motion detection turned off.
Solid red and green led's on my pi2

ccrisan, 2015-10-21:
Make sure your power adapter supplies enough current. Also please send me the logs as indicated here: https://github.com/ccrisan/motioneyeos/wiki/Reporting-An-Issue#attach-log-files.

scoobyd00, 2015-10-21:
Thanks I'll be back home tomorrow so I'll get those log files sent to you then.

scoobyd00, 2015-10-22:
Sent you an email with the logs. Thanks.

scoobyd00, 2015-10-28:
Ive swapped out all my hardware to try and fix the problem but still having the crashes... UNTIL
I swapped my raspberry pi 2 for another I was using for another project and it solved the issue, been running for over 48hrs with no crashes. So it looks like the PI was at fault, must be faulty.

scoobyd00, 2015-10-28:
Spoke too soon, I've accessed the web remotely earlier on, tried again and it looks like its crashed. Can't access the webpage or via SSH remotely.
It was fine for over 48hrs but I didnt access it remotely over this period
I wonder if this has something to do with it, remote access.
I'll do a powercycle when I get home and see how long it stays up without crashing when not accessing it remotely.

ccrisan, 2015-10-28:
This sounds a lot like a power-related problem. It shouldn't have anything to do with logging in remotely, except for, possibly, some minor extra current drawn when using the network adapter. What's your power supply rated at (voltage and current)?

scoobyd00, 2015-10-28:
It's 5v 2amp .. I've got a couple of these running separate pi's for different projects.. I'll try swapping it out for one of the others tomorrow and let you know how I get on.
Thanks.

ccrisan, 2015-10-29:
2 amps is usually more than enough so that shouldn't be the problem, unless you're using long thin cables.

scoobyd00, 2015-10-29:
Ive just done a power cycle and swapped the supply for one of my others. I wont logon remotely over 4g and see how long it stays up. I'll keep you posted.

scoobyd00, 2015-10-29:
Just checked and it had crashed, solid gree led.
Ive changed, power supply, raspberry pi, cables, im using a remote rtsp camera so dont have anything else connected to the PI.
Ive just trying a replacement SD card to see what happens with that.

ccrisan, 2015-10-30:
What exactly do you have connected to your PI and how exactly is it configured to work?

maxl95, 2015-10-31:
Hello,
i have the same problem as "scoobyd00". 
After four hours, it has been chrashed up. Was reached neither SSH nor via Web. 
I use a 5V 2Amp Powersupply and have only connected an USB flashdrive and a Webcam. 
Sorry for my bad english.

ccrisan, 2015-10-31:
@maxl95 can I see your logs?

maxl95, 2015-10-31:
Here are my logs.
https://drive.google.com/folder/d/0B03MQ-L2NMFOa2VXN3R1YmtWYTg/edit

scoobyd00, 2015-10-31:
Mines crashed again. Lasted 24hrs.. I've got nothing connected to my pi apart from Ethernet.
I'm using a rtsp camera which is connected to my router via wireless.
I'm writing to a USB drive which is also connected to my router. I have scheduled recording enabled and everything else is standard in the setup.

Ive ran Mrdaves fork of motion command line on my pi for weeks on end without any crashes so I'm sure it must be an issue with motioneyeos.
I've changed all my components, pi, power, SD card, cables and still have the same issue. I wonder if it could be rtsp related somehow?

scoobyd00, 2015-10-31:
Just seen that @maxl95 is using a USB webcam so doesn't seem its rtsp related.
Fingers crossed the issue can be tracked down because the software is excellent and a great frontend for motion

scoobyd00, 2015-10-31:
When I'm back home I'll try and use a USB camera connected to my pi instead of rtsp to see if that works without crashing to try and narrow down the issue

tyborall, 2015-11-01:
I have a similar problem with my raspberry pi 2. it crashes like you describe, sometimes after 1 hour or it runs stable 2 days until it crashes. I see that it crashed, because the WIFI stick LED is permanently lid. I use a WIFI connection only. Even with another WIFI stick (tried TN725 and Edimax) it crashes. With my other both Raspberry Pi 1 with motioneyeos I don't have such problems with the same WIFI sticks. I cannot find anything in the logs....

ccrisan, 2015-11-01:
Issues like this are almost impossible to debug, given that the system hangs without leaving any trace in the log files. Here are a few things that come to my mind:
 1. Swapping or removing components one by one until the issue goes away.
 2. Running the system with a monitor (and maybe a keyboard) connected for a couple of days so that you can actually see what's going on when it crashes.
3. Implementing a brown-out detection logging mechanism that will help us rule out power supply issues.

tyborall, 2015-11-01:
@scoobyd00 please check in expert settings, that your overclocking value is not set to "turbo". It seems that was the fault at my end. The 6 Volts should be the problem then. ;-)

ccrisan, 2015-11-02:
Thanks @tyborall, I updated my comment by adding the overclocking idea.

scoobyd00, 2015-11-02:
Ive just done a fresh SD card install.
Setup a monitor and keyboard to the PI.
Overclocking is as standard (medium)
Ill see how it goes now and report any issues.

jlecanda, 2015-11-02:
I am experiencing the same issue... systems locks completely at random times with no lights blinking and no SSH access. Sometimes it can last a few hours running and sometimes not even 30min.
I'm using a Raspberry Pi 2 with the CSI Cam module and wired Ethernet.
I've tried several settings in overclocking and GPU memory but still continues to freeze.
Happy to help providing any information to solve the issue.

tyborall, 2015-11-02:
I changed Micro SD Card from Samsung to SanDisk yesterday. Since then, I had no more problems. But I think, my overclocking setting was on turbo, so I cannot say if the root cause was either the SD card (compatibility issues) or the overclockig settings.

scoobyd00, 2015-11-02:
Ivd had no luck with anythjng.  Connected a monitor but it's just crashed after 7hrs, nothing but a blank unresponsive screen..
Last thing to try is a SanDisk sd card which ive just ordered.

ccrisan, 2015-11-03:
@scoobyd00 what are your overclocking settings? Can you post your /boot/config.txt file?

scoobyd00, 2015-11-03:
@ccrisan Im currently running motioneyeos on an old model b raspberry pi, still connected to my rtsp camera just to test to see what happens. So far its still up and running. Ive ordered a sandisk sd card for my raspberry pi 2 and will test that when it arrives as im sure the solid green light when it crashes is something to do with the write to sd card..

When ive tested the new sd card on the PI ill let you know what the current situation is.

bamoore01, 2015-11-06:
I have the same problem. Pi2 with a Sandisk card, wireless adaptor, 2 Amp supply.  Worked fine on the MotionPie software but locks up on the new version.  I haven't checked the over locking setting but will do that next.  

B

jlecanda, 2015-11-06:
I am no longer seeing this issue with the new version 20151103 and using the maximum GPU memory of 944. I am using the Pi2 overclocking setting (1000/500/500/2). Has been running for 2 days straight without crashing...  :)

scoobyd00, 2015-11-06:
I've been running the latest version as a test on my pi 2 and its been 48hrs now without a crash but I'm not using my rtsp camera due to the latest version not detecting the rtsp protocol, so just testing with an old USB cam

scoobyd00, 2015-11-08:
My pi 2 was running version 103 with a USB camera solid for 3 days.. Gone back to version 025 pre release as I wanted rtsp support back and its crashed within 12hrs.
Looks like it was an issue with the pre release versions.
Hopefully when rtsp support is available on the stable release it will fix the issues.,

scoobyd00, 2015-11-11:
Just an update and confirm, version 103 runs stable on my pi 2 with a USB camera (can't test rtsp yet but an update is coming) .
Version 025 crashes the pi 2 with the same USB camera or a rtsp camera

ccrisan, 2015-11-12:
That's great to hear. I'll close the issue then.

closed 2015-11-12 by ccrisan