motioneye-project/motioneyeos #11: Motionpie v.s. MotionEyeOS

duckietm, 2015-10-23:
Hello,

First of all thanks for this nice update, but after updateing 1 Rasbp2 (i have 5 ) i find the following :  

Please note that the setup + hardware are 100% same and i have both configured the overclocking to medium (950/250/450/0) they are both on 100Mbit /Full wired
I don't use recording through the motionPie /motionEyeOs, only got Streaming enabled so my QNAP can record this.

Rasp 2 with MotionPie :  

Camera : usb-Microsoft_Microsoft®_LifeCam_HD-3000-video-index0
Resolution: 800x448
Frame Rate: -- / 18
Bit Rate: 6.3 Mbps
Quality: --
Recording Status: Recording

MotionEyeOS :  

Camera : usb-Microsoft_Microsoft®_LifeCam_HD-3000-video-index0
Resolution: 800x448
Frame Rate: -- / 10
Bit Rate: 7.2 Mbps
Quality: --
Recording Status: Recording

If i change back the OS to MotionPie i get the right framerate again !
What is it i am doing wrong ???

ccrisan, 2015-10-23:
What are the framerate settings in your motionEye setup (the Video Device framerate and the Video Streaming framerate)?

duckietm, 2015-10-23:
I have the following setting
FPS 20 and Quality 83%

duckietm, 2015-10-23:
config : 

lightswitch 50
stream_authentication user:
snapshot_interval 0
ffmpeg_output_movies off
stream_motion off
target_dir /data/output
text_double off
height 448
stream_auth_method 0
width 800
stream_quality 83
threshold 9324
quality 85
post_capture 4
noise_level 31
on_event_end /usr/bin/python /usr/lib/python2.7/site-packages/motioneye/meyectl.pyc relayevent -c /data/etc/motioneye.conf -l stop %t
text_left
pre_capture 2
noise_tune on
picture_filename
locate_motion_style redbox
locate_motion_mode off
contrast 0
stream_maxrate 20
output_pictures off
hue 0
saturation 0
stream_localhost off
videodevice /dev/v4l/by-id/usb-Microsoft_Microsoft®_LifeCam_HD-3000-video-index0
ffmpeg_variable_bitrate 0
ffmpeg_video_codec msmpeg4
text_changes off
movie_filename %Y-%m-%d/%H-%M-%S
event_gap 10
auto_brightness on
stream_port 8081
rotate 180
text_right
on_event_start /usr/bin/python /usr/lib/python2.7/site-packages/motioneye/meyectl.pyc relayevent -c /data/etc/motioneye.conf -l start %t
brightness 0
ffmpeg_bps 2899999
framerate 30
max_movie_time 0
emulate_motion off
minimum_motion_frames 1
snapshot_filename

on the MotiionEyeOS

ccrisan, 2015-10-23:
What is the output of the command `uptime` while your streaming is in use? I'm interested in the system load.

duckietm, 2015-10-23:
uptime : 09:05:08 up 20 min,  load average: 0.51, 0.74, 0.55

Mem: 66152K used, 821060K free, 176K shrd, 2564K buff, 22356K cached
CPU:  11% usr   2% sys   0% nic  80% idle   0% io   0% irq   4% sirq
Load average: 0.54 0.74 0.55 3/77 2622

ccrisan, 2015-10-23:
The system load seems alright. What does "Frame Rate: -- / 10" mean? What do "--" and "10" stand for?

duckietm, 2015-10-23:
10 FPS

ccrisan, 2015-10-23:
The bitrate is higher with motionEyeOS so I really wouldn't know what to say here. Do you have other means of measuring the actual framerate just to make sure there's a performance regression with motionEyeOS?

duckietm, 2015-10-23:
the only 2 that takes up CPU : 

533   518 root     S    53692   6%  17% /usr/bin/motion -c /data/etc/motion.conf -n -d 1
  518     1 root     S    19848   2%   2% /usr/bin/python2.7 /programs/motioneye/motioneye.py --port 9000 --motion-binary /usr/bin/motion --log-level info --mjpg-client-idle-timeout 10 --add-remove-cameras true

ccrisan, 2015-10-23:
Well it's clearly not a CPU-related problem then. Maybe the camera driver has changed for the worse with the latest 4.x kernels available for the raspberry pi. Could you try using a different camera and repeating the scenarios?

duckietm, 2015-10-23:
Already did, and the Rasp camera works fine and also a logitech.

duckietm, 2015-10-23:
I don't know what happend but after 3 hours the camera start to run @ 19 FPS ???????

tyborall, 2015-10-25:
Already tried changing Motion Eye binary to /usr/bin/motion-mmal? I have better framerate with that.

duckietm, 2015-10-27:
yes that gives 3 FPS more , Thanks !

ccrisan, 2015-10-27:
Great! I'll close the issue then.

closed 2015-10-27 by ccrisan