motioneye-project/motioneyeos #12: [rpi2] Activating Fast Network Camera reboot and no more available

yar0d, 2015-10-23:
I've just installed version 20151021.  When I switch on Fast Network Camera, it reboot my Raspberry Pi 2 and then it became impossible to access it : Page Web inaccessible ERR_CONNECTION_REFUSED in Chromium. I've waited several minutes and tried several times and the result is same.

**boot.log**
```
---- booting ----
Setting hostname: done
Starting logging: done
Starting watchdog: done
Starting eudev: done
Setting up data partition
/dev/mmcblk0p3: Bad magic number in super-block while opening filesystem
Formatting data partition
Discarding device blocks:   4096/916608528384/916608             done                            
Creating filesystem with 916608 4k blocks and 229376 inodes
Filesystem UUID: 0db302a7-20a9-4478-a399-e5ffdf72bf95
Superblock backups stored on blocks: 
	32768, 98304, 163840, 229376, 294912, 819200, 884736

Allocating group tables:  0/28     done                            
Writing inode tables:  0/28     done                            
Creating journal (16384 blocks): done
Writing superblocks and filesystem accounting information:  0/28     done

Setting empty admin password
Starting wpa_supplicant: done
Starting network: wlan0: starting dhclient
eth0: flags=4098<BROADCAST,MULTICAST>  mtu 1500
        ether b8:27:eb:f7:b1:f7  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        loop  txqueuelen 0  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.0.244  netmask 255.255.255.0  broadcast 192.168.0.255
        ether 74:da:38:0c:c6:d5  txqueuelen 1000  (Ethernet)
        RX packets 8  bytes 1894 (1.8 KiB)
        RX errors 0  dropped 2  overruns 0  frame 0
        TX packets 5  bytes 1398 (1.3 KiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

done
ssh-keygen: generating new host keys: RSA1 RSA DSA ECDSA ED25519 
Starting sshd: done
Setting current date using http: Fri Oct 23 08:31:38 UTC 2015
Starting http date updater: done
Starting proftpd: done
Starting crond: done
Setting smb admin password: done
Starting smbd: done
Starting nmbd: done
Starting motioneye: done
---- shutting down ----
Stopping logging: done
Stopping watchdog: done
Stopping eudev: done
Shutting down wpa_supplicant: done
Stopping network: done
Stopping sshd: done
Shutting down date updater: done
Stopping proftpd: done
Stopping crond: done
[... 636 lines cut ...]
2015-10-23 08:37:02: [motioneye]    ERROR: streameye restart failed
2015-10-23 08:37:02: [motioneye]     INFO: adding new camera with id 1...
2015-10-23 08:37:03: [motioneye]    ERROR: HTTP 400: Bad Request (unknown operation)
2015-10-23 08:37:03: [motioneye]  WARNING: 400 GET /picture/1/current/?_=1445589423760&_username=admin&_signature=5b5fb8d0f6e582b60d160a9a7fd9e722aa51b54f (192.168.0.10) 14.98ms
2015-10-23 08:37:04: [motioneye]    ERROR: HTTP 400: Bad Request (unknown operation)
2015-10-23 08:37:04: [motioneye]  WARNING: 400 GET /picture/1/current/?_=1445589425236&_username=admin&_signature=83a3400885387271dec10a169d6178087aab9c52 (192.168.0.10) 5.82ms
2015-10-23 08:37:04: [motioneye]     INFO: rebooting
2015-10-23 08:37:04: [motioneye]     INFO: executing "/sbin/reboot"
2015-10-23 08:37:06: [motioneye]     INFO: interrupt signal received, shutting down...
2015-10-23 08:37:06: [motioneye]     INFO: server stopped
2015-10-23 08:37:07: [motioneye]     INFO: interrupt signal received, shutting down...
```

**raspimjpeg.log**
```
2015-10-23 08:42:15:  INFO: raspimjpeg.py 0.5
2015-10-23 08:42:15:  INFO: hello!
```

**streameye.log**
```
2015-10-23 08:42:14: INFO : streamEye 0.5
2015-10-23 08:42:14: INFO : hello!
2015-10-23 08:42:14: INFO : listening on 0.0.0.0:8081
```

ccrisan, 2015-10-23:
Thanks for the detailed report. I'll look into it.

yar0d, 2015-10-23:
SSH access is still working :-)

AlexandreBeauduin, 2015-10-24:
Same problem ! But SSH is not working here :)

snagglewest, 2015-10-24:
Same result here as the other report.

ccrisan, 2015-10-24:
I have identified and fixed the bug. I have yet to rebuild the images and I'll release the new version these days.

ccrisan, 2015-10-25:
I prepared a new prerelease that should fix the fast network camera issue. Please check it out and let me know if the problem persists.

snagglewest, 2015-10-27:
I just upgraded about a hour ago but it seems to be working as expected in fast network mode!

ccrisan, 2015-10-27:
Thanks for the feedback. I'll close the issue then.

closed 2015-10-27 by ccrisan