motioneye-project/motioneyeos #14: Usb CAM + email notifications problem

anandbhala, 2015-10-25:
Hi,

I'm a newbie..trying to explore motioneyeos.  It is awesome as far as the capabilities of it are.  But I'm having a few problems

1. USB Camera crashes with grey image saying unable to open video device since.....
---Initially when it happened, I modified the config.txt with max_usb_current=1 now it is a bit more stable, but crashes after sometime, but a reboot works

here is the output of motion -n

------------------------
cap.driver: "sonixj"
cap.card: "USB camera"
cap.bus_info: "usb-bcm2708_usb-1.4"
cap.capabilities=0x85200001
------------------------
[1] [NTC] [VID] v4l2_get_capability: - VIDEO_CAPTURE
[1] [NTC] [VID] v4l2_get_capability: - READWRITE
[1] [NTC] [VID] v4l2_get_capability: - STREAMING
[1] [NTC] [VID] v4l2_select_input: name = "sonixj", type 0x00000002, status 00000000
[1] [NTC] [VID] v4l2_select_input: - CAMERA
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1] [NTC] [VID] v4l2_set_pix_format: Config palette index 17 (YU12) doesn't work.
[1] [NTC] [VID] v4l2_set_pix_format: Supported palettes:
[1] [NTC] [VID] v4l2_set_pix_format: (0) JPEG (JPEG)
[1] [NTC] [VID] v4l2_set_pix_format: 0 - JPEG (compressed : 1) (0x4745504a)
[1] [NTC] [VID] v4l2_set_pix_format Selected palette JPEG
[1] [NTC] [VID] v4l2_do_set_pix_format: Testing palette JPEG (352x288)
[1] [NTC] [VID] v4l2_do_set_pix_format: Using palette JPEG (352x288) bytesperlines 352 sizeimage 51278 colorspace 00000007
[1] [NTC] [VID] v4l2_scan_controls: found control 0x00980900, "Brightness", range 0,255
[1] [NTC] [VID] v4l2_scan_controls:     "Brightness", default 128, current 128
[1] [NTC] [VID] v4l2_scan_controls: found control 0x00980901, "Contrast", range 0,127
[1] [NTC] [VID] v4l2_scan_controls:     "Contrast", default 20, current 20
[1] [NTC] [VID] v4l2_scan_controls: found control 0x00980902, "Saturation", range 0,40
[1] [NTC] [VID] v4l2_scan_controls:     "Saturation", default 25, current 25
[1] [NTC] [VID] v4l2_scan_controls: found control 0x0098090e, "Red Balance", range 24,40
[1] [NTC] [VID] v4l2_scan_controls:     "Red Balance", default 32, current 32
[1] [NTC] [VID] v4l2_scan_controls: found control 0x0098090f, "Blue Balance", range 24,40
[1] [NTC] [VID] v4l2_scan_controls:     "Blue Balance", default 32, current 32
[1] [NTC] [VID] v4l2_scan_controls: found control 0x00980910, "Gamma", range 0,40
[1] [NTC] [VID] v4l2_scan_controls:     "Gamma", default 20, current 20
[1] [ERR] [VID] v4l2_set_mmap: Error requesting buffers 4 for memory map. VIDIOC_REQBUFS: Device or resource busy
[1] [NTC] [VID] vid_v4lx_start: Using V4L1
[1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[1] [NTC] [VID] vid_close: Closing video device /dev/video0

2.  Email notifications are not being sent...it says connection terminated unexpectedly.  I saw your pre-release which says it is fixed, but I still receive the same problem.  

Can you please help me with this?

Thank you,
Anand

ccrisan, 2015-10-25:
Make sure your power adapter can supply the required current.

s3nox, 2015-10-26:
I'm having similar problems with a Logitech C920 webcam. MotioneyeOS gives me a grey screen and message "unable to open video device". SSH into RPI and try to issue command v4l2-ctl -L comes back with no such device or directory. This happens with in ten minutes.

ccrisan, 2015-10-26:
Yeap, a limited current can have this effect of disconnecting the USB devices.

closed 2015-10-26 by ccrisan