motioneye-project/motioneyeos #16: How to Update

vdyo, 2015-10-27:
Hi, I have the first pre-release installed, is there an easy way to update it, or do I need to re-image?

Regards

Damian

ccrisan, 2015-10-28:
Just enable Prereleases in Expert Settings and hit the update button.

vdyo, 2015-10-28:
Hi, I just tried that and I get a The update process has failed! error. I added enable pre release and I got the error when I chose Update, I restarted for good measure but same error.

ccrisan, 2015-10-28:
Can I see your motionEye.log?

vdyo, 2015-10-28:
sure one sec

vdyo, 2015-10-28:
```conf_path /data/etc
run_path /tmp
log_path /var/log
media_path /data/output
motion-binary /usr/bin/motion-mrdave
log-level info
listen 0.0.0.0
port 80
mount_check_interval 300
motion_check_interval 10
cleanup_interval 43200
thumbnailer_interval 60
remote_request_timeout 10
mjpg_client_timeout 10
mjpg-client-idle-timeout 10
smb_shares true
smb_mount_root /data/media
wpa_supplicant_conf /data/etc/wpa_supplicant.conf
local_time_file /data/etc/localtime
enable_reboot true
enable_update true
smtp_timeout 60
zip_timeout 500
add_remove_cameras true```

ccrisan, 2015-10-28:
The motioneye.log file, please (it's downloadable from the UI).

vdyo, 2015-10-28:
ah ok

vdyo, 2015-10-28:
Hi, I just emailed it to you. Thanks Damian

ccrisan, 2015-10-28:
You log ends on the 14th of October and there's no sign of updating in it. Have you run out of space on your SD card?

vdyo, 2015-10-28:
hmm 14.3/14.3 GB (99%) I guess so, let me see how to delete stuff.

ccrisan, 2015-10-28:
Well that's your problem. After that, the update process should work flawlessly. Feel free to reopen this issue if it doesn't.

closed 2015-10-28 by ccrisan