motioneye-project/motioneyeos #17: No image in email

vdyo, 2015-10-28:
Hi

Firsty I just sent you a new log, upgrade went fine and I get email notifications, but no image.

Regards

Damian

ccrisan, 2015-10-28:
"ERROR: timeout waiting for the media listing process to finish" indicates that you probably have too many media files and motionEye decided to send you the notification email empty instead of waiting even longer for the pictures to be retrieved. The solution is to clean up a bit your media files folder; parsing them on an SD card can be quite slow.

vdyo, 2015-10-28:
Hi, how do I clean it up? I did delete everything that was in it. Should I use a USB device instead and make the video and stills directory?

ccrisan, 2015-10-28:
Approximately how many media files (pictures and videos) do you currently have associated to the camera (use the buttons on the top bar of your camera frame). As long as you keep let's say no more than an a few hundreds of pictures, you should be fine with a common SD card.

vdyo, 2015-10-28:
Sorry my mistake I wasnt looking at the images just the video...there is 20K of them..I'll delete and sort out the setting to only keep for 7 days. I'll test and come back to you shortly.

vdyo, 2015-10-28:
Hi, Ok I had the Frame Change Threshold set to low, so it was grabbing 10 images very quickly, which is why the files were not ready to be sent when there was a notification. I reduced it and I just got my first email with the frame. So All working well now. Great System again!

closed 2015-10-28 by vdyo