motioneye-project/motioneyeos #18: Anyone have a solution to sync with DropBox

vdyo, 2015-10-28:
I saw the google Drive one, is there a similiar method?

oopeekoo, 2015-10-29:
There is this: https://github.com/andreafabrizi/Dropbox-Uploader
It works fine in Raspibian (done a quick test once), haven't tried it in motioneyeos or motionpie.

IamC8, 2015-11-14:
Hi look here: https://github.com/ccrisan/motioneyeos/issues/31 The problem is the stat command. The code requires BIN_DEPS="sed basename date grep stat dd mkdir". All are present but not stat.

vdyo, 2015-11-23:
Thanks for the answer, might be a bit my own Linux ability, but I can probably find someone to add it for me.

ccrisan, 2015-11-23:
"Official" support for Google Drive and Dropbox is coming soon. Stay tuned :)

vdyo, 2015-11-23:
Awesome

SuneBielefeldt, 2015-11-23:
Dropbox and Google Drive support
It's very good news, looking forward to the next great version.
😊

closed 2015-11-24 by vdyo