motioneye-project/motioneyeos #19: WiFi dongle causes repeated (endless) loops

caveflyer, 2015-10-30:
Installed version 20151011, and **everything works very well using an ethernet cable**.  After following all instructions on enabling WiFi, shutting down, inserting WiFi dongle (I've tried 2 different ones, **both that work with Raspbian**), the Pi2 goes through a lot of the boot process, but when it gets to starting motionpie, it restarts.  The last 2 lines I see are:

```
Starting motioneye: done
[   36.111257] watchdog stopped
```

then the process starts all over and ends the same way, with a restart.  If I remove the dongle, and go back to using the ethernet cable, everything works fine.

(I tried changing the DHCP to static; no change.  I was very, very careful in typing in my SSID and password -- no cut-n-paste -- but that isn't the issue either.  As I said, these two WiFi dongles worked well with Raspbian.)

motionpie.log ::::

```
2015-10-30 04:32:46: [motioneye]     INFO: hello! this is motionEye server 20151011
2015-10-30 04:32:46: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2015-10-30 04:32:46: [motioneye]     INFO: cleanup started
2015-10-30 04:32:46: [motioneye]     INFO: wsswitch started
2015-10-30 04:32:46: [motioneye]     INFO: thumbnailer started
2015-10-30 04:32:47: [motioneye]     INFO: server started
2015-10-30 04:32:47: [motioneye]     INFO: adding new camera with id 1...
2015-10-30 04:32:48: [motioneye]  WARNING: control hue not found for device /dev/video0
2015-10-30 04:32:48: [motioneye]     INFO: main config file /data/etc/motion.conf does not exist, using default values
2015-10-30 04:32:50: [motioneye]  WARNING: control hue not found for device /dev/video0
2015-10-30 04:33:20: [motioneye]    ERROR: HTTP 404: Not Found (not found)
2015-10-30 04:33:20: [motioneye]  WARNING: 404 GET /nice%20ports%2C/Tri%6Eity.txt%2ebak (192.168.0.119) 4.12ms
2015-10-30 04:33:20: [motioneye]    ERROR: HTTP 405: Method Not Allowed
2015-10-30 04:33:20: [motioneye]  WARNING: 405 OPTIONS / (192.168.0.119) 3.96ms
2015-10-30 04:33:50: [motioneye]  WARNING: control hue not found for device /dev/video0
2015-10-30 04:33:50: [motioneye]  WARNING: control hue not found for device /dev/video0
2015-10-30 04:33:50: [motioneye]  WARNING: control hue not found for device /dev/video0
2015-10-29 23:34:51: [motioneye]     INFO: rebooting
2015-10-29 23:34:51: [motioneye]     INFO: executing "/sbin/reboot"
2015-10-29 23:34:52: [motioneye]     INFO: interrupt signal received, shutting down...
2015-10-29 23:34:52: [motioneye]     INFO: server stopped
2015-10-29 23:34:56: [motioneye]     INFO: motion stopped
2015-10-29 23:34:56: [motioneye]     INFO: SMB shares unmounted
2015-10-29 23:34:56: [motioneye]     INFO: bye!
2015-10-29 23:34:56: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 104] Connection reset by peer
2015-10-29 23:34:56: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 104] Connection reset by peer
1969-12-31 18:00:11: [motioneye]     INFO: hello! this is motionEye server 20151011
1969-12-31 18:00:14: [motioneye]     INFO: cleanup started
1969-12-31 18:00:14: [motioneye]     INFO: wsswitch started
1969-12-31 18:00:14: [motioneye]     INFO: thumbnailer started
1969-12-31 18:00:14: [motioneye]     INFO: server started
1969-12-31 18:00:35: [motioneye]    ERROR: authentication failed for user admin
1969-12-31 18:00:43: [motioneye]  WARNING: control hue not found for device /dev/video0
1969-12-31 18:00:43: [motioneye]  WARNING: control hue not found for device /dev/video0
1969-12-31 18:00:44: [motioneye]  WARNING: control hue not found for device /dev/video0
1969-12-31 18:01:19: [motioneye]  WARNING: control hue not found for device /dev/video0
1969-12-31 18:01:19: [motioneye]  WARNING: control hue not found for device /dev/video0
1969-12-31 18:01:30: [motioneye]     INFO: shutting down
1969-12-31 18:01:30: [motioneye]     INFO: executing "/sbin/poweroff"
1969-12-31 18:01:31: [motioneye]     INFO: interrupt signal received, shutting down...
1969-12-31 18:01:31: [motioneye]     INFO: server stopped
1969-12-31 18:01:34: [motioneye]     INFO: motion stopped
1969-12-31 18:01:34: [motioneye]     INFO: SMB shares unmounted
1969-12-31 18:01:34: [motioneye]     INFO: bye!
1969-12-31 18:01:34: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 104] Connection reset by peer
1969-12-31 18:01:34: [relayevent]    ERROR: failed to relay event: [Errno socket error] [Errno 104] Connection reset by peer
1969-12-31 18:00:29: [motioneye]     INFO: hello! this is motionEye server 20151011
1969-12-31 18:00:32: [motioneye]     INFO: cleanup started
1969-12-31 18:00:32: [motioneye]     INFO: wsswitch started
1969-12-31 18:00:32: [motioneye]     INFO: thumbnailer started
1969-12-31 18:00:32: [motioneye]     INFO: server started
1969-12-31 18:00:34: [motioneye]     INFO: interrupt signal received, shutting down...
1969-12-31 18:00:34: [motioneye]     INFO: server stopped
1969-12-31 18:00:37: [motioneye]     INFO: motion stopped
1969-12-31 18:00:37: [motioneye]     INFO: SMB shares unmounted
1969-12-31 18:00:37: [motioneye]     INFO: bye!
[... 906 lines cut ...]

done
Starting sshd: done
Setting current date using http: failed
Starting http date updater: done
Starting proftpd: done
Starting crond: done
Setting smb admin password: done
Starting smbd: done
Starting nmbd: done
Starting motioneye: done
---- shutting down ----
Stopping logging: done
Stopping watchdog: done
Stopping eudev: done
Shutting down wpa_supplicant: done
Stopping network: done
Stopping sshd: done
Shutting down date updater: done
Stopping proftpd: done
Stopping crond: done
Shutting down smbd: done
Shutting down nmbd: done
Shutting down motioneye: done
```

ccrisan, 2015-10-30:
Thanks for the detailed report. I can see that `Starting wpa_supplicant: failed` shows up in your log, indicating that... well... the wpa_supplicant daemon has failed. Here are a few questions:

1. The boot.log file does not make any reference to a wlan0. Given that boot.log is overwritten with every boot, did you capture it while having your wifi adapter plugged in?
2. What wireless encryption do you use? motionEyeOS only supports WPA.
3. Do you have any special characters in your SSID or PSK? Or maybe a hidden SSID?

caveflyer, 2015-10-30:
Thank you for the quick response!

Question 2:  I used WPA2 on my Netgear C6300.  I've tried both 2.4Ghz and
5Ghz, but no difference in end result.  (My dongles are only 2.4G, but I
had to try something!)

Question 3: No special characters, just upper-, lower-case alphabetic
characters and numbers, no spaces.  I've been very careful about this,
since I saw it mentioned several times in your other discussions.

As for question #1, I'd be happy to try whatever it takes to capture the
boot log, but since the Pi2 reboots on its own and the only way for me to
get to it is after I re-plugin the ethernet cable and reboot, I'm not sure
how to do that.

Thanks again!

  -- Robert

ccrisan, 2015-10-30:
What wifi adapter model do you use? The driver (or the firmware) may not be included in the distro.

caveflyer, 2015-10-30:
I have two...  one came with the Canakit I got, and the other is the EDIMax.

I first loaded Raspbian and made the changes necessary to make WiFi work,
and it did work.  Then I used another microSD with MotionEye on it, and it
has not worked.

  -- Robert

ccrisan, 2015-10-30:
Both of them worked with Raspbian and none of them works with motionEyeOS?

caveflyer, 2015-10-30:
Well, I only tried the Canakit WiFi, and it worked with Raspbian.  I didn't
try the other one with Raspbian.   I have, however, earnestly tried both of
them with MotionEye.  I first made sure it was powered off, and then I
inserted each (at different times, of course) and then powered on.  The
Canakit dongle would never turn on it's LED as the Pi2 cycled over and
over.  With the EDIMax, the blue LED would slowly flash at various times as
the Pi2 cycled, but it never gave any indication that is was
sending/receiving (i.e., no fast flashing).

caveflyer, 2015-10-30:
I figured it out!  And I feel bad.  In *most* places I've used my SSID, it
was "Netgear33".  I noticed that in one, and only one, place was it shown
as "NETGEAR33".  I had been using the former, so I tried the latter (all
uppercase) and it worked!

Sorry to have wasted your time!  And thanks for the great support!

  -- Robert

ccrisan, 2015-10-31:
No worries, I'm glad you solved the problem.

closed 2015-10-31 by ccrisan

tsrgit, 2016-02-25:
I am having a similar problem, but not sure if I should continue here or open a new issue.  I can setup and configure my RPi B using the 20160221 release of motioneyeos.  I configure wifi with the ssid and password of my router (which does not broadcast its ssid).  When I reboot, the RPi goes into an endless reboot loop and never fully boots.  To get the system to boot, I have to pull out the usb wifi dongle and use a wired ethernet connection.  I used the exact same hardware with a release of motionpie from about a year ago and had no problems with wifi.  Attached are my log files.

[logs.zip](https://github.com/ccrisan/motioneyeos/files/146056/logs.zip)

ccrisan, 2016-02-25:
@tsrgit your logs don't indicate a software-induced reboot, but rather a hardware related one. Could you try using a powered USB hub and see if the problem persists? It might be due to the extra power drawn by the wifi adapter when connected.

tsrgit, 2016-02-26:
I tried using a powered USB hub, but nothing changed.  Infinite reboot loop.

ccrisan, 2016-02-26:
@tsrgit could you please temporarily connect a monitor and see if anything suspicious is displayed?

tsrgit, 2016-02-27:
I connected a monitor and wpa_supplicant failed on boot.  I tried moving the RPi to another room and it booted fine!  So the problem appears to be my wifi signal.  The weird thing is that the wifi connection failed when the RPi was a foot away from my wireless router, but then worked when it was in a different room.  Go figure.

ccrisan, 2016-02-29:
@tsrgit could you please remove the file `/etc/modprobe.d/modprobe.conf` and see if the problem persists? Also please paste here the output of the command `lsmod`.

tsrgit, 2016-03-04:
@ccrisan I figured it out.  The problem was with my wireless router.  I have a couple of them set up as wireless access points, and one of them had crashed.  So I had no wireless signal on one side of my house, but still had signal on the other side.  Booting the RPi up a foot away from the router did me no good because the router was actually down!  And when I moved the RPi to the other side of the house, it worked of course.  So all is well.  Thanks for sticking with me on this, even though motioneye was completely blameless.

ccrisan, 2016-03-04:
Alright, it's great to hear that you solved the problem.