motioneye-project/motioneyeos #20: Cannot write: No space left on device

csolg, 2015-10-30:
Hi,

I cannot rebuild image.

Thanks.

```
$ ./build.sh raspberrypi2        
>>>   Finalizing target directory
rm -rf /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/lib/luarocks
mkdir -p /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/etc
echo "primcam.ru" > /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/etc/issue
mkdir -p /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/etc/network/
( echo "# interface file auto-generated by buildroot"; echo ; echo "auto lo"; echo "iface lo inet loopback"; ) > /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/etc/network/interfaces
rm -rf /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/include /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/share/aclocal \
	/home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/lib/pkgconfig /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/share/pkgconfig \
	/home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/lib/cmake /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/share/cmake
find /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/{lib,share}/ -name '*.cmake' -print0 | xargs -0 rm -f
find /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/lib \( -name '*.a' -o -name '*.la' \) -print0 | xargs -0 rm -f
find /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/lib \( -name '*.a' -o -name '*.la' \) -print0 | xargs -0 rm -f
rm -rf /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/share/gdb
rm -rf /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/share/zsh
rm -rf /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/man /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/share/man
rm -rf /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/info /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/share/info
rm -rf /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/doc /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/share/doc
rm -rf /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/share/gtk-doc
rmdir /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/share 2>/dev/null
Makefile:560: recipe for target 'target-finalize' failed
make[1]: [target-finalize] Error 1 (ignored)
find /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target -type f \( -perm /111 -o -name '*.so*' \) -not \( -name 'libpthread*.so*' -o -name '*.ko' \) -print0 | xargs -0 /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/host/usr/bin/arm-buildroot-linux-gnueabihf-strip --remove-section=.comment --remove-section=.note 2>/dev/null || true
if test -d /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/lib/modules; then \
	find /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/lib/modules -type f -name '*.ko' -print0 | \
	xargs -0 -r /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/host/usr/bin/arm-buildroot-linux-gnueabihf-strip --remove-section=.comment --remove-section=.note --strip-unneeded; fi
find /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/lib -type f -name 'libpthread*.so*' | \
	xargs -r /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/host/usr/bin/arm-buildroot-linux-gnueabihf-strip --remove-section=.comment --remove-section=.note --strip-debug
mkdir -p /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/etc
# Mandatory configuration file and auxiliary cache directory
# for recent versions of ldconfig
touch /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/etc/ld.so.conf
mkdir -p /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/var/cache/ldconfig
if [ -x "/home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/host/usr/bin/arm-buildroot-linux-gnueabihf-ldconfig" ]; \
then \
	/home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/host/usr/bin/arm-buildroot-linux-gnueabihf-ldconfig -r /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target \
				-f /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/etc/ld.so.conf; \
else \
	/sbin/ldconfig -r /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target \
		       -f /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/etc/ld.so.conf; \
fi
/sbin/ldconfig.real: Warning: ignoring configuration file that cannot be opened: /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/etc/ld.so.conf: No such file or directory
/sbin/ldconfig.real: /usr/lib/libstdc++.so.6.0.19-gdb.py is not an ELF file - it has the wrong magic bytes at the start.

( \
	echo "NAME=Buildroot"; \
	echo "VERSION=2015.08-rc1-gb771222"; \
	echo "ID=buildroot"; \
	echo "VERSION_ID=2015.08-rc1"; \
	echo "PRETTY_NAME=\"Buildroot 2015.08-rc1\"" \
) >  /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/etc/os-release
>>>   Copying overlay board/common/overlay
>>>   Copying overlay board/raspberrypi2/overlay
>>>   Executing post-build script board/common/postscript.sh
rm: cannot remove ‘/home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/usr/share/perl5’: Is a directory
>>>   Generating root filesystem image rootfs.ext2
rm -f /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/build/_fakeroot.fs
rm -f /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target/THIS_IS_NOT_YOUR_ROOT_FILESYSTEM
rm -f /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/build/_users_table.txt
echo "set -e" >> /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/build/_fakeroot.fs
echo "chown -h -R 0:0 /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/target" >> /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/build/_fakeroot.fs
cat system/device_table.txt > /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/build/_device_table.txt
printf '	/bin/busybox                     f 4755 0  0 - - - - -\n 	/var/lib/nginx d 755 33 33 - - - - -\n /dev/console c 622 0 0 5 1 - - -\n' >> /home/old/csolg/Downloads/linux/motioneyeos/motioneyeos/output/raspberrypi2/build/_device_table.txt
[... 1077 lines cut ...]
tar: ./etc/ssl/certs/Sonera_Class_1_Root_CA.pem: Cannot create symlink to ‘../../../usr/share/ca-certificates/mozilla/Sonera_Class_1_Root_CA.crt’: No space left on device
tar: ./etc/ssl/certs/DigiCert_High_Assurance_EV_Root_CA.pem: Cannot create symlink to ‘../../../usr/share/ca-certificates/mozilla/DigiCert_High_Assurance_EV_Root_CA.crt’: No space left on device
tar: ./etc/ssl/certs/AffirmTrust_Premium_ECC.pem: Cannot create symlink to ‘../../../usr/share/ca-certificates/mozilla/AffirmTrust_Premium_ECC.crt’: No space left on device
tar: ./etc/ssl/certs/USERTrust_RSA_Certification_Authority.pem: Cannot create symlink to ‘../../../usr/share/ca-certificates/mozilla/USERTrust_RSA_Certification_Authority.crt’: No space left on device
tar: ./etc/ssl/certs/Verisign_Class_3_Public_Primary_Certification_Authority_-_G2.pem: Cannot create symlink to ‘../../../usr/share/ca-certificates/mozilla/Verisign_Class_3_Public_Primary_Certification_Authority_-_G2.crt’: No space left on device
tar: ./etc/ssl/certs/GeoTrust_Universal_CA.pem: Cannot create symlink to ‘../../../usr/share/ca-certificates/mozilla/GeoTrust_Universal_CA.crt’: No space left on device
[... cut ...]
tar: ./etc/ssl/certs/NetLock_Business_=Class_B=_Root.pem: Cannot create symlink to ‘../../../usr/share/ca-certificates/mozilla/NetLock_Business_=Class_B=_Root.crt’: No space left on device
tar: ./etc/ssl/certs/Camerfirma_Chambers_of_Commerce_Root.pem: Cannot create symlink to ‘../../../usr/share/ca-certificates/mozilla/Camerfirma_Chambers_of_Commerce_Root.crt’: No space left on device
tar: ./etc/ssl/certs/Security_Communication_RootCA2.pem: Cannot create symlink to ‘../../../usr/share/ca-certificates/mozilla/Security_Communication_RootCA2.crt’: No space left on device
tar: Exiting with failure status due to previous errors

```

ccrisan, 2015-10-30:
You probably added some relatively large packages to the image which is set to a limited size of 180 MB for the root partition and 20 MB for the boot partition. If you really need these large packages on your OS image, you could edit `board/common/mkimage.sh` and change ROOT_SIZE and DISK_SIZE to something bigger.

csolg, 2015-10-30:
Thank you! It works.

closed 2015-10-30 by csolg