motioneye-project/motioneyeos #21: video encoding in hardware

ivr588, 2015-10-31:
Hi,
Would motioneyeos be able to take advantage of enabling mpeg-2 hardware encoding as per http://swag.raspberrypi.org/collections/software/products/mpeg-2-license-key?

many thanks,

ivr588, 2015-10-31:
Thought I'd also mention that https://github.com/ccrisan/motioneyeos/issues/90 which is the Recommended Configurations section off https://github.com/ccrisan/motioneyeos/wiki/Configuration gives a 404.

ccrisan, 2015-10-31:
I'd need to do some research before being able to answer your question. Thanks for letting me know about the dead link.

ivr588, 2015-11-01:
Hi,

[pi2 with piNoIR camera]

I got the mpeg-2 licence anyway. and yeah it does make a difference for the better. Not so laggy and it's running 10fps with ease, and not crashing. Before, if I had everything running (streaming, movies and stills), even at the 448 maximum GPU sometimes it would lag and crash. Now it has 944 MB for GPU (modified boardctl.py and rebuilt), running sshfs mounted output dir to remote site and it's stable so far.

Having it as just a mjpeg camera 800x480 @15fps quality=100 gives really smooth results.

tyborall, 2015-11-03:
I modified thread-1.conf for testing purposes, removed the "videodevice" line, pointed to network camera http://127.0.0.1:8080 with "netcam_url" and launched raspimjpeg.py piped to streameye. Performance is even better. If we now write the videos motion creates to a ram disk and after that copy it to sdcard, i think we can achieve 25 fps @ 1920x1080. 
The project https://github.com/tyborall/raspi_security_camera inspired me...

ccrisan, 2015-11-03:
Unfortunately it's not as easy as it seems. Simply using the netcam way instead of the v4l2 way won't improve the performance at all. On the contrary, the overhead of locally creating the stream (using streameye) will slightly degrade the performance.

The good idea is that we could supply motion with a lower resolution/quality feed so that the motion detection doesn't eat up that much CPU. This is an idea to be further developed and the implementation is for sure not that easy.

tyborall, 2015-11-03:
That way Raspi_Security_Camera works. It uses motion on a very low resolution and it's only for triggering the video recording after motion is detected. But maybe it's an approach to work on.

tyborall, 2015-11-05:
Hey Calin, have you ever looked in PiKrellCam? http://billw2.github.io/pikrellcam/pikrellcam.html
The web interface is ugly, but the feature set is stunning! Real 30 FPS Full HD recording and everything works without "motion" package. It is very very stable. 

Maybe you can adapt that c written daemon for use with the CSI camera board.

ektus, 2016-01-16:
I've got both PiKrellCam and MotionEye running under Raspbian Wheezy on a PI2. PiKrellCam takes care of the internal camera module, set to 10fps at 1296*972 and video recording on motion detect, works good with very few false detects. 
MotionEye is set to do motion detection for a RTSP network camera, but currently only on the preview stream (320x240 with MotionEye set to 640x480), but produces many false motion detects due to compression artefacts (entire regions of the picture becoming garbled). I don't know if this is a problem of the camera, the network or the RPi, though :-(

bortek, 2017-02-23:
Closing this due to lack of interest.

closed 2017-02-23 by bortek