motioneye-project/motioneyeos #22: Unable to install noip for dynamic DNS setup

Witblits, 2015-11-02:
I refer to the instructions given by http://pimylifeup.com/raspberry-pi-port-forwarding/

The "make" command is missing and thus I am unable to create a binary from the "noip" source files. I tried installing "make", but it seems there is no packet manager such as apt-get available in this OS. I also tried to directly use the binaries supplied by the noip folder, but none of them are compatible for the Raspberry Pi 2 architecture. No I'm thinking of compiling the source file on a virtual machine identical to that of the rpi2, but thought I might come for assistance first. Thanks in advance.

maxl95, 2015-11-02:
Hello Witblits,
you can use the update URL of No-IP together with a python-script to update your DDNS. If everything works fine, you can put the script into an crontab. 
That's the way i do it.

This is the URL:
http://username:password@dynupdate.no-ip.com/nic/update?hostname=mytest.testdomain.com&myip=127.0.0.1

Witblits, 2015-11-05:
Hi maxl95
Thanks for the great suggestion! I did as you said. However now I realized that my router does not assign a consistent local IP address to my RPi2 (the problem being that the port forwarding then still uses the old local IP address). I could not find a functionality on the router webpage to allocate a static local IP address. So I would like to assign the local IP address by editing the /etc/network/interfaces file on the RPi2, however the file does not exist and creating the file from scratch with the appropriate inputs also does not solve the problem. I'm not sure if this MotionEyeOS has the interfaces file stored somewhere else or if there is another workaround to this problem?

Thank you again in advance!

maxl95, 2015-11-05:
Normally you can find this in the menu " Network " . Habe a look at my picture.

[image]

closed 2017-02-23 by bortek

bodait, 2024-06-25:
> Hello Witblits, you can use the update URL of No-IP together with a python-script to update your DDNS. If everything works fine, you can put the script into an crontab. That's the way i do.
> 
> This is the URL: http://username:password@dynupdate.no-ip.com/nic/update?hostname=mytest.testdomain.com&myip=127.0.0.1

Hello there,
I'm sorry, but i do not understand how to achieve this. I don't know where the mentioned python script is, how to use it together with the "update url" (https://www.noip.com/support/knowledgebase/ddns-settings-request-url) and where (crontab) to put that script. Would you be kind to give the less experienced step-by-step instructions?
Your help would be much appreciated!
Tamas

bodait, 2024-07-10:
Well, i was in contact with NO-IP support staff, and after quite a few round of emails we had a solution. Just run the NO-IP DUC on another machine on your local newtwork. If it's possible for you, this is perhaps the simplest solution.