motioneye-project/motioneyeos #23: protocol rtsp not supported - release 20151103

scoobyd00, 2015-11-04:
With the latest release when trying to add a rtsp network camera im unable to do so and
get the error 'protocol rtsp not supported'.

The pre-release versions worked fine.

scoobyd00, 2015-11-07:
@ccrisan is it easy to sort this issue? The latest release had been running stable on my pi 2 for over 48hrs now but without rtsp support

ccrisan, 2015-11-08:
The issue has been fixed. I'll release a new version within the following week.

scoobyd00, 2015-11-08:
Excellent thanks.

closed 2015-11-08 by scoobyd00

dmolner, 2015-11-28:
Realy work?
I test but with same message 'protocol rtsp not supported'.

ccrisan, 2015-11-28:
I haven't gotten to release the new version yet. Hang on another few days.

dmolner, 2015-11-28:
Perfect! 
With this option I only need two more, upload ftp and take pictures when motion occur.

scoobyd00, 2015-12-02:
New release is working well with rtsp.
Thankyou.