motioneye-project/motioneyeos #26: Motion not recording

stuartbloom, 2015-11-09:
Hi there

Thanks very much for an A1 pieced of code, VERY impressive.  I have all installed, however, when I tried to install ffmpeg I get the error:

Package ffmpeg is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'ffmpeg' has no installation candidate

This is installing on the latest Raspian.

Another thing is, that although I can see the camera border turn red, no files are being written to the folder.  Can anyone help?  Also, is it possible to create sub folders for each camera and then for year\month\day within that folder?

Do you have any time frame for dropbox integration?  Will you be also looking at box.com?

Thanks again, and sorry if these are silly questions

Stuart

ccrisan, 2015-11-09:
If you use motionEye on a recent Raspbian, you need to follow [these recently updated instructions](https://github.com/ccrisan/motioneye/wiki/Install-On-Raspbian). More precisely, step 1 will help you add a third party repo where you should be able to find the ffmpeg package.

stuartbloom, 2015-11-09:
hmm, thanks for that; I have run step 1, rebooted the pi, but am still not seeing any files in the /var/lib/motioneye folder :(

ccrisan, 2015-11-09:
Please send me your thread-1.conf file and your motion.log file.

stuartbloom, 2015-11-09:
Seem to be unable to attach here, so sending files to your email; hope that's OK.

Stuart

ccrisan, 2015-11-09:
Well as far as I can tell from your config file, you haven't enabled movies.

stuartbloom, 2015-11-09:
oh, how embarrassing, I am so sorry. please accept my apologies.  time to go and lie down in a darkened room.

Thanks
for pointing out my stupidity ;)

Stuart

ccrisan, 2015-11-09:
No worries.

closed 2015-11-09 by ccrisan