motioneye-project/motioneyeos #27: Move /data to USB

psts, 2015-11-09:
How would I go about moving /data to a USB-stick?

I tried the regular route by changing /etc/fstab's /data to /dev/sda1, but that didn't work.

ccrisan, 2015-11-10:
The fstab way should normally work, although I haven't tested it. Why would you want to do that anyway? The data partition stores two big pieces of information: (1) the custom configuration and (2) the media files. If you want your media files to be stored on your USB stick, you could do that from the UI.

psts, 2015-11-16:
The reason is because I compile MotionEyeOS from source because I have custom resolutions and a few other things. That way I could have /data with all the settings (and images/videos) saved to a USB-stick and then just pop out the SD-card and overwrite it when ever there is a new version.

ccrisan, 2015-11-16:
You can already do that with the current motionEye version. Just configure the root directory for the media files to be the USB stick. Then, when you want to rewrite the SD card image (for whatever reason), download the entire configuration using the "backup" button and restore it when you have written and booted the new image.

closed 2015-11-16 by ccrisan