motioneye-project/motioneyeos #28: email troubles...

sinbrkatetete, 2015-11-10:
Hi. Sorry to be bothering you, but I'm having troubles with email notifications. The last time they worked for me was on the last version of motionpie. I'm suspecting it has something to do with "from" field in the email being sent. Did you perhaps change something in that sense and could you please tell me if it's possible for a noob to set it back the way it was (editing a config file?) or would you be willing to set it back (if it proves to be the cause of problems). Here are two messages from different mail hosts I've found in the log:

1. - smtp.mail.ru
line 731, in sendmail
SMTPSenderRefused: (550, 'not local sender over smtp'

2. - mail.gmx.com
line 731, in sendmail
SMTPSenderRefused: (550, 'Requested action not taken: mailbox unavailable\nSender address is not allowed.

I'm suspecting that gmail "quietly" changes the from field and lets the message through, but these two don't, they return an error.

Thanks in advance!

ccrisan, 2015-11-12:
The `From` field has been recently set to the destination `To` address, because normally you're sending an email to yourself and because there were some users reporting that their SMTP servers refuse to work otherwise.

What are your settings and what would your SMTP server accept as `From`?

sinbrkatetete, 2015-11-12:
Ahaa! I get it. 

Short description of my setup:
Two locations (house1, house2), each has it's own account @ mail.ru (house1@mail.ru, house2@mail.ru). Each location has devices that use the corresponding house account to send various alerts (voicemail/fax,MotionEyeOS) to a couple of recipients on different email services.
MotionEyeOS is set up to use house1/house2 accounts (depending on the location) @ mail.ru to send alerts to two email addresses on different email service providers. The outgoing server is smtp.mail.ru, port 587, TLS ON. The recipients' addresses differ from the sender's (recipient1@gmail.com and recipient2@gmx.com) on both of the Raspberries. When the MotionEyeOS tries to send an email via smtp.mail.ru, the email gets rejected because the "from" field isn't the same as the account name (house1@mail.ru / house2@mail.ru).
As far as I can tell, the smtp.mail.ru (and mail.gmx.com) server would like the "from" field to be the same as the account name that's being used to send the email (for house1 - it should be house1@mail.ru, for house 2 - house2@mail.ru i.e. the from field should, in my case, be the same as the login name to the outgoing server used), not the addresses of the recipient(s). BTW, which email address is used as the "from" address when there are two or more addresses specified as recipients?

I don't know anything about mail servers, so I don't know if there are some that need only the username as the login or do they all use "username@service.extension" as the login string, but if they all (can) use "username@server.extension" variant, would it be too much to ask to set the default behavior to use the sender account username@service.extension as the "from" field? That way, the users who send the emails to the sender email address (themselves) wouldn't be / shouldn't be affected. Or, if you're worried it would break the service for other (majority of?) users, would it be too much to ask for a step-by-step instruction on how to change the behavior by myself? If both of the ideas are too much to ask of you, I just remembered I could try and set the receiver=sender and try and set up filtering/forwarding rules on mail.ru to forward the messages to the real recipients.

Thanks for all you're doing!

ccrisan, 2015-11-13:
Indeed it's the first in the To list that is used as From. Is this workaround good enough for you or do you think I should consider adding an explicit option for the From field?

sinbrkatetete, 2015-11-13:
It's more than good enough for me... Thank you very much for asking. 

As to putting an option for defining the sender ("from" field) in the gui... Well, if it's not too much work for you, maybe you should add such an option. It's mere existence could point anyone having the same troubles as me to the right solution and, in my opinion, it's better to cover all the points of eventual problems when you encounter them so you don't have to think about it in the future, when this suite grows. On the other hand, I have no idea how much people could/will encounter this problem in the future... but still, the "from field" option shouldn't be a point of confusion, especially if you put a short explanation under the question mark tip - something like "input email address you'd like to appear as the sender, beware - some providers like it to be the same as the full name of the account you're using to send the messages".

Once again, thanks for building this, even more thanks for taking the time to deal with me and my troubles. You've made a great piece of software and I hope it grows into something that'll provide some financial benefit for you besides the learning opportunities it already did/does. 

All the best in coding and life!

bortek, 2017-02-23:
@sinbrkatetete Is this still current or can we close the issue?

closed 2017-03-01 by bortek