motioneye-project/motioneyeos #30: How to save snapshots/videos on shared folder on network

JoseAlvaradoRuiz, 2015-11-11:
Hello

I am quite new to the Motion Eye OS (great work btw!), and my problem relays on the save path for the pictures/videos.
I shared a folder (an external hard drive in a laptop, with all the permisions, withouth username and password) on my network, and when i try to configure the file storage as a "Network Share" with the information os the shared folder, it seems that the Motion Eye doesn't recognize the path (no video displays), altough i can access the shared folder from other computers.

Am I doing something wrong? Have I not done something before sharing?

Attaching an image for a better reference

Thanks in advance

[image]

ccrisan, 2015-11-12:
There's something wrong with your share configuration (either on the server side or on motionEyeOS setup side). I couldn't tell what the problem is, I suggest using IP addresses instead of machine names (what's "MyNetworkOrIP" btw?). Also make sure the name of the share is indeed "MyFolder" and that it's indeed guest-enabled (no user/password).

JoseAlvaradoRuiz, 2015-11-12:
Thanks for the answer.

"MyNetworkOrIP" was just something i wrote for the screenshot. I configured that with my laptop name and "MyFolder" was indeed the share name.
I just tried again the configuration, now including my username and password of my laptop and it worked!
Thanks a lot :)

closed 2015-11-12 by ccrisan