motioneye-project/motioneyeos #31: stat Command

IamC8, 2015-11-14:
Hi, I notice in the source that you use coreutils 8.24. This webpage says taht the command 'stat' is included? When I try to use it it complains about '-sh: stat: command not found'. If it is not in the build can you please tell me how to get it in. The reason is I want to use this page to get Dropbox running. http://raspi.tv/2013/how-to-use-dropbox-with-raspberry-pi

Thank you for all your hard work.

ccrisan, 2015-11-14:
The next version will have the stat command included.

IamC8, 2015-11-15:
Hi ccrisan. I fixed it by recompiling the source and adding coreutils, thanks. That script for Dropbox is just soooo slow, its not worth it. Thanks in any case, someone might need it in the future still.
Just have a look at coreutils.mk ,the line COREUTILS_BIN_PROGS = cat chgrp chmod chown cp date dd df dir echo false ln ls mkdir mknod mv pwd rm rmdir vdir sleep stty sync touch true uname join

stat is not in there also. I very new to this and inserted it here also. Not sure if that is necessary or not.

Also the site http://www.mpfr.org is down, I got mpfr-3.1.3.tar.xz from here http://ftp.gnu.org/gnu/mpfr/

ccrisan, 2015-11-15:
Actually, for motionEyeOS, many of the coreutils are provided by busybox. `stat` is one of them and I have enabled it now.

closed 2015-11-15 by ccrisan