motioneye-project/motioneyeos #33: Raspberry PI USB WiFi dongle not supported

agassner, 2015-11-20:
Hi,

I'm getting an error running on a Raspberry Pi 2 Model B with the official WiFi dongle - https://www.raspberrypi.org/products/usb-wifi-dongle/
Error:
```
usb 1-1.3: Direct firmware load for brcm/brcmfmac43143.bin failed with error -2
```
It seems that the firmware is not there.

I tested on Raspbian and it works fine. The firmware is at /lib/firmware/brcm/

Can you help on this?

Thanks for the brilliant work, well done!

Arthur

ccrisan, 2015-11-20:
I'll make sure to include the required firmware in the next release.

agassner, 2015-11-20:
Nice one, thanks.

ccrisan, 2015-11-21:
The firmware has been added to the OS. It will be available with the next release.

closed 2015-11-21 by ccrisan