motioneye-project/motioneyeos #34: Solid white speckle

psts, 2015-11-21:
Now when the snowy season has started, false triggers is becoming a hassle.

Snow, rain and bugs that come too close to the sensor, are all white. People, cars and other things that are far away, are never all white.

So, is there a way to "cancel out" solid white speckle" that triggers the motion detection?

ccrisan, 2015-11-21:
[This](http://www.lavrsen.dk/foswiki/bin/view/Motion/ConfigOptionDespeckle) is what could be done regarding your issue. To be honest, it's beyond my expertise and I have no idea what values would be suitable. Feel free to try the despecle option by adding it to the extra motion options.

psts, 2015-11-21:
I have already played with that. The problem is the "speckle" is the same size as a person, so I can't increase the threshold and noise level doesn't seem to change anything. Only difference is the color.

Not sure what to do.

ccrisan, 2015-11-21:
In that case I don't think there is anything to do about it. At least not with *motion*. I'll close the issue now and mark it as won't fix, as I really wouldn't know how to fix it :)

closed 2015-11-21 by ccrisan