motioneye-project/motioneyeos #35: Add remote motioneye cam

SuneBielefeldt, 2015-11-22:
I have tried to add second MotionEyeOS / RPi2 to the first MotionEyeOS / RPi2. Unfortunately, I receive an connection error. Is it because it is not possible to add Remote MotionEyeOS units to a unit already using its egt camera. 

To be used in a stand alone MotionEyeOS RPi2 in order to function as a hub MotionEye? Maybe it's just my URL syntax is wrong, but the device tries to connect, so I must believe it is correct. 

Can you briefly describe the requirements for a MotionEyeOS hub / RPi2. 
How many cameras will be able to handle? 
Being shown streaming tumbs from each camera? 

It's a great fine piece of software you have developed here. Great respect and gratitude for it. 

Craving someone design a 3D printable waterproof encloure to house it all.

Kind regards
Sune Bielefeldt
Denmark

ccrisan, 2015-11-23:
What is the URL that you're using to add the first motionEye to the second motionEye? Normally everything should work fine as long as the versions of the two units are the same.

SuneBielefeldt, 2015-11-23:
I use the following url: http://192.168.1.108:8081 and get connection error: request time out

Before that i have remove all other camera connections.

Se picture at dropbox link:
https://www.dropbox.com/sh/ouuryr2bta0zken/AABwbKLNbDY1zu_CJdkPBAgla?dl=0

ccrisan, 2015-11-23:
Use the port 80, as that's where motionEye lives (http://192.168.1.108). Port 8081 is associated with the MJPEG streaming, which, when requested, never ends the response (thus giving a timeout when used as you did).

SuneBielefeldt, 2015-11-24:
Dear Calin Crisan 

Have tried your instructions to connect a MotionEyeOs Remote Camera with http://192.168.1.108:80, but here too, the following error: Fail to connect to 192.168.1.108 port 80, no host to route. 

During connection is shown an example of the screen which is suggested as follows: 
Example: http: //examble.com: 8080/cam ... also here given an error: connection refused. 

See my dropbox link for screenshots from my experiments.
(https://www.dropbox.com/sh/ouuryr2bta0zken/AABwbKLNbDY1zu_CJdkPBAgla?dl=0)

SuneBielefeldt, 2015-11-24:
Hello Calin Crisan 

Sorry, I had used a large H in the Http: // and it was evidently here the fault lay. 

Now everything works as it should and I'm glad and happy. 

Thinking that you might want to correct the example to show that to be used port 80 instead of 8080. 

Thank you for your amazing work with MotionEyeOs, it is a super project and the great work you put into it.

See my dropbox link for latest screenshots from my experiments with succes.
(https://www.dropbox.com/sh/ouuryr2bta0zken/AABwbKLNbDY1zu_CJdkPBAgla?dl=0)

SuneBielefeldt, 2015-11-24:
Thanks a lot

closed 2015-11-24 by SuneBielefeldt

ccrisan, 2015-11-24:
Where exactly do you think I should correct the example?

SuneBielefeldt, 2015-11-24:
Dear Calin Crisan.

On the screen "add camera" which indicate url to Remote MotionEye Camera says
"http://example.com: 8080/cams. The correct must be "http://example.com:80". 

I think it could be a great help when you, as inexperienced in MotionEyeOs first time trying to add a remote cam.

Sune

ccrisan, 2015-11-24:
Yes, you're right. Thanks.

SuneBielefeldt, 2015-11-24:
You are welcome 😊

ioquatix, 2017-09-10:
This is a good suggestion and I was confused by it too.

spyda96, 2019-02-21:
The easy way to add the other motion eye cam was just to copy the ip address from the other cam and past it into the gui. This works everything for me.