motioneye-project/motioneyeos #36: Cant quite get Motion Detection and email working

vdyo, 2015-11-23:
Hi

I am still trying to get motion detection and email working, I have tried various settings, but never seem to capture the image of the person for example walking up the the camera, can anyone share their better set up info for the camera and motion detection?

I get a lot of ```Motion has been detected by camera "Driveway/meye-12b5579b" at 2015-11-23 19:04:35 (Australia/Brisbane).``` with no attachments as well.

Thanks

Damian

ccrisan, 2015-11-23:
Have you enabled still images? If yes, what Capture Mode have you selected. What value have you set for Attached Pictures Time Span when you configured email notifications?

vdyo, 2015-11-23:
I did and some come through, I'll send you what I have set.

vdyo, 2015-11-24:
Hi

I have set.

Show Frame Changes: ON
Frame Change Threshold: 5%
Auto Noise Detection: ON
Light Switch Detection: 50%
Motion GAP: 10 Deconds
Captured Before: 100 Frames
Captured After: 100 Frames
Mon Motion Frames: 100

Still Images Settings.
Image Quality: 100%
Capture Mode: Motion Triggered
Preserve Pictures: For One Week

I will get an email 10 minuted apart, with no image, then some will have an image, they might be an hour apart.

ccrisan, 2015-11-24:
Those relatively big numbers related to frames don't look good. Try captured before/after and min motion frames with smaller values (e.g. 5). Check to see if still images are actually saved on your disk around the moment of the motion event.

You never answered my question regarding the Attached Pictures Time Span setting.

josejnet, 2015-12-11:
Same problem here.

Show Frame Changes: ON
Frame Change Threshold: 5%
Auto Noise Detection: ON
Light Switch Detection: 25%
Motion GAP: 10 Deconds
Captured Before: 5 Frames
Captured After: 5 Frames
Mon Motion Frames: 5

Still Images Settings.
Image Quality: 75%
Capture Mode: Motion Triggered
Preserve Pictures: For One Week

Pictures Time Span setting: 30S.

What should I change? Working under wifi.

ccrisan, 2015-12-12:
@josejnet what exactly is your problem? Does your system detect motion at all? Have you seen the camera frame turning red?

josejnet, 2015-12-12:
hey @ccrisan thank you. 

I have two Rp1. The first one with Pi Noir working under cable. The second one Pi Cam working under wifi. 
Both have exactly the same settings. The the one working under wifi is not attaching pictures to mail notifications.
More info:
The camera frame turns red.
It is taking pictures and saving them into the sd card.
it is uploading pictures to Google Drive. 

But still not pictures attached to mail. Right now the settings are

Motion GAP: 10 Deconds
Captured Before: 2 Frames
Captured After: 4 Frames
Mon Motion Frames: 10
Attached Pictures Time Span 30 seconds.

The one over ethernet cable works great.

ccrisan, 2015-12-12:
And do you actually see some pictures taken around the moment of a motion event (in the media browser)? If yes, do you have a large number of pictures?

josejnet, 2015-12-12:
it took 6 pictures in 6 seconds. Thanks,

ccrisan, 2015-12-12:
So does it work or not?

josejnet, 2015-12-12:
Sorry, no. Not pictures attached yet.

ccrisan, 2015-12-12:
Can I see your motion.log and your motioneye.log files?

josejnet, 2015-12-12:
motion.log
using motion binary /usr/bin/motion-mmal
[-1244949640] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[-1244949640] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[-1244949640] [NTC] [ALL] motion_startup: Motion mmaltest Started
[-1244949640] [NTC] [ALL] motion_startup: Logging to syslog
[-1244949640] [NTC] [ALL] motion_startup: Using default log type (ALL)
[-1244949640] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD

josejnet, 2015-12-12:
motioneye.log

18:09:13: [motioneye]    ERROR: failed to upload file "/data/2015-12-12/18-08-29.jpg" with service gdrive: 'str' object is not callable
2015-12-12 18:09:20: [motioneye]  WARNING: Connect error on fd 22: ECONNREFUSED
2015-12-12 18:10:01: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-12 18:10:04: [sendmail]     INFO: email sent
2015-12-12 18:11:49: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-12 18:11:53: [sendmail]     INFO: email sent
2015-12-12 18:14:19: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-12 18:14:23: [sendmail]     INFO: email sent
2015-12-12 18:26:38: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2015-12-12 18:26:49: [motioneye]  WARNING: Connect error on fd 17: ECONNREFUSED
2015-12-12 18:27:20: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-12 18:27:23: [sendmail]     INFO: email sent
2015-12-12 18:40:21: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2015-12-12 18:40:32: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2015-12-12 18:41:14: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-12 18:41:17: [sendmail]     INFO: email sent
2015-12-12 18:43:12: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2015-12-12 18:43:22: [motioneye]  WARNING: Connect error on fd 15: ECONNREFUSED
2015-12-12 18:43:22: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-12 18:43:27: [sendmail]     INFO: email sent
2015-12-12 18:45:24: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2015-12-12 18:45:24: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-12 18:45:29: [sendmail]     INFO: email sent
2015-12-12 18:45:33: [motioneye]  WARNING: Connect error on fd 15: ECONNREFUSED
2015-12-12 19:08:33: [sendmail]    ERROR: timeout waiting for the media listing process to finish
2015-12-12 19:08:45: [motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed
2015-12-12 19:08:53: [sendmail]    ERROR: failed to send mail: [Errno -3] Temporary failure in name resolution
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 194, in on_message
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 51, in send_mail
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
  File "/usr/lib/python2.7/smtplib.py", line 316, in connect
  File "/usr/lib/python2.7/smtplib.py", line 291, in _get_socket
  File "/usr/lib/python2.7/socket.py", line 557, in create_connection
gaierror: [Errno -3] Temporary failure in name resolution
                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                                  2015-12-12 19:13:55: [motioneye]     INFO: hello! this is motionEye server 20151206
2015-12-12 19:13:59: [motioneye]     INFO: cleanup started
2015-12-12 19:13:59: [motioneye]     INFO: wsswitch started
2015-12-12 19:14:00: [motioneye]     INFO: tasks started
2015-12-12 19:14:00: [motioneye]     INFO: mjpg client garbage collector started
2015-12-12 19:14:00: [motioneye]     INFO: smb mounts started
2015-12-12 19:14:00: [motioneye]     INFO: server started
2015-12-12 19:17:18: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2015-12-12 19:17:27: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED

ccrisan, 2015-12-12:
The line `[sendmail] ERROR: timeout waiting for the media listing process to finish` indicates that there are just too many media files in your media folder and they cannot be parsed fast enough so that the email be sent within a reasonable time. The solution is to clean up a bit your media folder.

closed 2015-12-12 by ccrisan

onshisan, 2017-04-22:
I'm not sure if my problem is similar to this one or not....

Motion notifications (with attached images) were working just fine for me when I first set it up this morning, but now I am not getting any emails. Motion does seem to be detected, and images are being saved. When I test my (gmail) email settings, I get the "[Errno -3] Temporary failure in name resolution". I did not adjust any email settings between the successful and failing states. I did, however, try to turn on movies (but found that none were being recorded). Turning off movies has not resolved this issue, so now I am without motion notifications.

Any suggestions are much appreciated.

Thanks for this excellent software, by the way! If I can get this up and running I have big plans :)

ccrisan, 2017-04-23:
*Temporary failure in name resolution* indicates a DNS configuration error. Make sure your system has access to the internet and that it can resolve domain names. This usually happens when users set an invalid static IP configuration. Also double check your SMTP server address.

onshisan, 2017-04-23:
Static IP does seem to be the issue. It's funny, I can access the web interface to administer motioneye both on my LAN and over the internet (including from my mobile phone over LTE) but can no longer browse the web on the Ubuntu box that's running motioneye. 

I had set up a static IP via my router's DHCP server (back when things were working fine), and it continues to take that IP when I reboot the machine, but my router is no longer showing this machine in its DHCP client list, even when the router is rebooted with the static IP reserved for the motioneye server's mac address is deleted from the router's table.

I'll play with the network settings on the Ubuntu machine a little bit and see if I can resolve it, otherwise I may just wipe it any start over.