motioneye-project/motioneyeos #38: No video on multi cams with higher resolution.

woale, 2015-11-23:
Been running MotionEyeOS with a single 720p cam on a Pi2 for some time, and lovin' it ;)

Now added a second 720p webcam, but video simply stops (no cam icon) as soon as resolution reaches 640x480 (and upwards) even with lowest framerate setting on both cams... though 1280x720 on 1st, and 320x240 on 2nd cam (or vice versa) works fine with high framerate.

Playing with settings (text/streaming/stills/movies/motion disabled) doesn't seem to help.
When video stops, Motioneye.log contains lots of following warnings (x being a number);
"[motioneye]  WARNING: Connect error on fd x: ECONNREFUSED"

Any suggestions in troubleshooting ? 
What are the known limitations (terms of number of cams, resolution, framerate) running a Pi2 ?

Thanks for any help!
Woale

ccrisan, 2015-11-23:
The limitations are hard to determine, especially given that there are too many possible combinations (different boards, different camera types with different resolutions and framerates).

Nevertheless, in your case there might be a problem of available power. Please send me the dmesg.log file and try it with a better power supply.

woale, 2015-11-23:
Wow, impressive response time Calin ;)

Tried a few power supplies (ranging from 0.8A to 2A), same result.

Attached log file:
[dmesg.log.txt](https://github.com/ccrisan/motioneyeos/files/41489/dmesg.log.txt)

ccrisan, 2015-11-23:
Can I see all the logs in fact (notably motion.log and motioneye.log)?

woale, 2015-11-23:
Sure, I've just removed some 404 GET Warnings from motioneye.log

[motion.log.txt](https://github.com/ccrisan/motioneyeos/files/41731/motion.log.txt)
[motioneye.log.txt](https://github.com/ccrisan/motioneyeos/files/41732/motioneye.log.txt)

woale, 2015-11-23:
Note, I'm getting 'inconsistent' behaviour.... now both cams are capable of showing resolution in 800x448 at 2-3fps.

ccrisan, 2015-11-23:
Hmm, the motion.log file is almost empty, as though the motion daemon didn't live too much.

Here are the important bits of your motioneye.log:
`2015-11-14 15:13:43: [motioneye]    ERROR: [Errno 2] No such file or directory: '/var/log/motion.log'`: usually happens when either you haven't given the unit enough time at first boot to create the necessary stuff on the data partition, or the sdcard got corrupted (/var/log/ is in fact a link to the /data/log, on the data partition).

Later on we see another similar complaint: `2015-11-14 15:26:53: [motioneye]    ERROR: failed to execute statvfs: [Errno 2] No such file or directory: '/data/output'`.

Moving on, there's this: `2015-11-15 19:38:09: [motioneye]  WARNING: 404 GET http://zc.qq.com/cgi-bin/common/attr?id=260714&r=0.6499879148173627 (115.230.124.164) 8.55ms`. You have probably forwarded port 80 from a public IP address, which is not a good idea, given the random HTTP scans that happen all the time. It will only kill the motionEye HTTP server.

On the other hand, I'm seeing lots of `2015-11-23 15:54:21: [motioneye]    ERROR: motion not running, starting it` in your motioneye.log. This happens when motion crashes. And motion probably crashed because of previous errors, in your case.

My recommendations are as follows:
 1. Use the more powerful PSU (you're probably using the 2A adapter now, but corruptions might have occurred while using the less powerful one).
 2. Try a new SD card and rewrite the OS image, from scratch.
 3. When booting the new system, never manually cut the power or reboot during the first boot.
 4. Use a different port for accessing it (e.g. 8888 -> 80).
 5. Keep me posted.

woale, 2015-11-26:
Followed your recommendations, and get slightly different behavior... whereby now one cam shows a grey video with the message "Unable to open video device" as soon as resolution is set to 640x480 (and up), while running with a single cam can get up to 720p 30fps.

Logs attached:
[motioneye.log.txt](https://github.com/ccrisan/motioneyeos/files/45241/motioneye.log.txt)
[motion.log.txt](https://github.com/ccrisan/motioneyeos/files/45239/motion.log.txt)
[dmesg.log.txt](https://github.com/ccrisan/motioneyeos/files/45240/dmesg.log.txt)

I'll try to find a different second cam (other then the Logitech C270) to test.

ccrisan, 2015-11-26:
Your motion.log file says the following:

    [1] [ERR] [VID] VIDIOC_TRY_FMT failed for format v4l2_set_pix_format: Input/output error
    [1] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
    [1] [NTC] [VID] vid_v4lx_start: Using V4L1
    [1] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device

I believe there might be a problem with the driver. Are the two devices identical? The driver might make at some point the assumption that there's only one device of that kind. It's just a wild guess, though.

dimovdimcho, 2015-12-28:
Calin, what is the best resolution you were able to achieve with a single camera module?
Where do you think the limitations lie when it comes to achieving the maximum 5MP resolution?

ccrisan, 2015-12-30:
I've been using my cameras at 1280x1024@10fps, obviously in Fast Network Camera mode. When it comes to FNA I believe there's a tradeoff between the JPEG quality, the resolution and the frame rate. More precisely as long as you keep the general data rate somewhere around 4MB/s, your PI1 should do fine. Also the FNA accepts FullHD resolution (1920x1080).

If you must use the motion daemon (i.e. FNA disabled), then the limitations come from the motion daemon itself which needs to do a lot of processing and is a lot less lightweight than streamEye (used for FNA streaming). I'm not familiar with motion's internals and I have no idea whether there are things to improve or not. All I know is that 5MP is not realistic when it comes to motionEye(OS), regardless of the camera or the board.

bortek, 2017-02-23:
Since the user did not response I consider this case resolved and will close it for now.

closed 2017-02-23 by bortek