motioneye-project/motioneyeos #40: Central server and multiple RPis

rafi1975, 2015-11-24:
Hi,

I just saw a scenario on the wiki page showing multiple RPis connected to a central server. The question is - how to get motioneyeos installed on e.g. CentOS server (not another RPi) so I have a hub there where all my RPis stream to?
Can you provide instruction if such scenario is achievable?

Thanks,

Rafal

ccrisan, 2015-11-24:
Just install [motionEye](https://github.com/ccrisan/motioneye/wiki/Installation).

rafi1975, 2015-12-03:
Just did that on dedicated CentOS box Iyou have to install all possible python packages) and configured network share. All flying - 6 PIs stream to that box with no issues so far.

Thanks,

Rafal

ccrisan, 2015-12-03:
That's great. Can we consider this issue closed then?

johndcrumpton, 2015-12-04:
I have a related question. I have 4 PIs, all with motioneyeos installed. I also have a central server with Ubuntu and motionEye. I can connect the Pis to the server either as remote motion or fast network camera. As fast network, there doesn't seem to be anyway to capture a video? Perhaps I misunderstood, but I thought the PI wouldn't be able to capture w/ the camera in this mode, but I thought the server would be able to write the PI's stream to the server's local storage?
When connected as remote motion cameras, I can capture, but it only seems to capture to the remote PIs, not the server. The only way I can see around that is to use a network share to put the files on the server from the PI... but if I'm going that route, what function does motioneye on the server serve?

ccrisan, 2015-12-04:
Use Fast Network Camera on your 4 PIs and add them as Network Cameras on your server. Attach the storage device to your local server, so that the motion daemon running on the server can save the media files locally, directly on your storage device.

rafi1975, 2015-12-04:
But this will disable motion detection functionality ?

ccrisan, 2015-12-04:
Why would it?

rafi1975, 2015-12-04:
I found that one the other day:

'label': 'Fast Network Camera',
491	        'description': 'Enabling this option will turn your Raspberry PI into a simple and fast MJPEG network camera, ' +
492	                'disabling motion detection, media files and all other advanced features (works only with the CSI camera)',

I was confused with - disabling motion detection.

ccrisan, 2015-12-04:
Well yes, indeed, when enabling Fast Network Camera on a RPi, that RPi will no longer detect motion or have other advanced features; it will turn into a simple but fast network camera, just like any other network camera on the market. If you add it to your central server as a Network Camera, your central server will do the motion detection and everything else.

closed 2015-12-04 by ccrisan

rafi1975, 2015-12-04:
All clear now.
Thanks for you prompt reply, as usual. Excellent job with that software!!!
I think we can close that ticket.

CrazyEyesPete, 2016-04-14:
Thank you for this answer, I was a little confused too.

kekkokk, 2019-07-28:
Maybe it's a stupid question but is there any way to add the pi camera to the server?
I mean, I'd like to configure my rpi camera so when it boots it auto register to the main server. this is because my server is on the cloud and my cameras are connected with private ips in my local home netwrok. I cannot reach them from my server but my server is reachable from my cameras.
Instead I saw that I have to add my cameras from the server and it seems to be a bit stupid