motioneye-project/motioneyeos #42: Setting up WiFi with MotionEye

rockman55, 2015-11-24:
[bootlog.txt](https://github.com/ccrisan/motioneyeos/files/42963/bootlog.txt)
[dmesglog.txt](https://github.com/ccrisan/motioneyeos/files/42962/dmesglog.txt)
[messageslog.txt](https://github.com/ccrisan/motioneyeos/files/42960/messageslog.txt)

[motionlog.txt](https://github.com/ccrisan/motioneyeos/files/42961/motionlog.txt)

Hello Calin, I have upgraded to the new OS and having some issues setting up my system using WiFi. I can get everything loaded using a direct Ethernet connection and I change all the settings to use the correct WiFi SSID but cannot find the correct login. I tried to change the /data/etc/wpa_supplicant.conf file to the SSID i use on other applications, cannot find the system. I am enclosing all the log files associated with this project, hopefully you will be able to tell me the error of my ways.
Thank,
Danny
PS Trying to remane the files as .txt to send them here.

ccrisan, 2015-11-24:
It seems that your WiFi connection is working just fine. Your boot log says: `Starting wpa_supplicant: done` and then you get the IP address `192.168.1.42` via DHCP. So I'd say everything is just fine.

I don't understand your problem. What do you mean by "cannot find the correct login"? What do you mean by "cannot find the system"?

rockman55, 2015-11-24:
Calin I am sending you a new boot log. With this configuration I have my WiFi adapter connected and no Ethernet cable connected. My router says that the ip address for MotionEye is 192.168.1.42. However, when I try to connect to that IP with my browser I get “The page cannot be displayed”. I have issues with this Edimax adapter usually when I try and set a static-ip address, the adapter usually wants to make the ip address different. When I know what ip the adapter wants I change the conf file to match that IP address. I am not sure how to use the command line to check for the ip address with MotionEye, the command I use with my other Pi’s is hostname –I and that does not work here. Please see the new boot.log file I used for the WiFi test.

 

Thanks,

Danny

 

PS my comments from the previous email are somewhat ambiguous , mostly what I was trying to say was that I could not log in to the web page.

ccrisan, 2015-11-25:
The IP address is not linked to your PI, but to the WiFi adapter or to the Ethernet port. You need to find your IP address for the WiFi adapter.

rockman55, 2015-11-25:
Ok how can I check that using MotionEye?
Thanks

https://github.com/ccrisan/motioneyeos/issues/42#issuecomment-159594477

ccrisan, 2015-11-25:
Usually your router tells you that. If not, connect a monitor and look for the ip address at the end of the boot messages.

rockman55, 2015-11-25:
I have done all of that and I see the same IP number, but still cannot connect to the web site for MotionEye. When I use this same adapter for a normal Raspbian software  and setup the wpa_supplicant.conf file with a random static ip, after booting I issue command hostname -I I get two different IP address displayed. To correct this I then change my static IP address to the one that was not previously in the wpa_supplicant.conf.
Why does this command (hostname -I) not work in MotionEye?
Thanks for your help
Danny

https://github.com/ccrisan/motioneyeos/issues/42#issuecomment-159672177

ccrisan, 2015-11-25:
1) The hostname command included with motionEyeOS is a stripped down version; therefore some functionalities may not be available. You can check your IP address using `ip addr`.
2) The `wpa_supplicant.conf` contains no IP address. I'm not sure how you added/changed an IP address in that file.
3) Does your PI respond to ping requests?

rockman55, 2015-11-25:
When I use the command ip addr, it shows that the IP address is 192.168.1.42, but I cannot ping that ip number (Destination host unreachable). I set up the same adapter into another system and it comes back with the same ip address 192.168.1.42 and I can ping that one. When I check for the MotionEye in my router status it comes up with the same ip address but still cannot access it through the web.

Thanks,

Danny 

 

I didn’t mean I was changing the IP address in the wpa_supplicant.conf file but I was setting up my SSID and password there.

ccrisan, 2015-11-26:
In that case, my only suggestion is to rewrite the OS image from scratch and start over with the configuration.

rockman55, 2015-11-26:
Thanks Calin, I did that and after some minor tweaking did get it to work. If one wanted to use WiFi instead of wired Ethernet, does MotionEye default to which connection is initially installed? When I had originally ran the program I had it wired and tried to change to WiFi and think maybe that is what started all my problems. Any comments are appreciated.
Thanks,
Danny

https://github.com/ccrisan/motioneyeos/issues/42#issuecomment-159859487

ccrisan, 2015-11-26:
If the wifi connection has been configured, the system will reboot until that connection is established. Ethernet, on the other hand, will be used only if the cable is connected.

closed 2015-11-26 by ccrisan

rsetala, 2016-01-21:
If the wifi has been configured (but that network is missing at the moment)  I don't get the ethernet link working with a cable connected?

ccrisan, 2016-01-22:
No, you don't. motionEyeOS is not Debian or a general purpose distro: if you configure it to work with WiFi, it will reboot indefinitely until it gets WiFi.

SulaimaniKeeda, 2016-11-01:
MotionEyeOS is great, works like a dream, I am a complete newbie...I have managed to have a video file save on to Dropbox and IFTTT takes over from there on to tell me of a motion being detected, I am using this to monitor my cat.
I am not able to access the page from outside local network. How does port forwarding work. 
Could use dataplicity on motion earlier but not on motioneyeos
I want to be able to access motioneye from outside home network.

ccrisan, 2016-11-01:
I'm sorry but port forwarding has nothing to directly with motionEyeOS and is something that you'll need to check with the documentation of your router. Please seek help on websites that deal with your router model or alike.

Kishiro80, 2017-06-02:
Can I ask , How to make MotioneyeOS be able to connect to wifi ? which fine should i change and what is the coding ?