motioneye-project/motioneyeos #44: Setting Up Email Object and Text

DaddyBabbes, 2015-11-25:
Hello and Thanks so much for your great work !!!
Is there a way to modify email template such as Object and body text ?

ccrisan, 2015-11-25:
No, there isn't.

Rodneykool, 2016-12-14:
Yes you can.

1. Connect with SSH, login with root and your password and make all the MotionEyeOS files writeable by running the following commands:
mount -o remount,rw /
mount -o remount,rw /boot

2. Connect with WinSCP (SSH) to the MotionEyeOS and download the "sendmail.pyc" file (location: /usr/lib/python2.7/site-packages/motioneye).

3. De-compile the sendmail.pyc file with for example "Easy Python Compiler" (https://sourceforge.net/projects/easypythondecompiler/)

4. Now you can edit all the email body and subject text in the python file.

5. Re-compile the sendmail.py file with Python by running: python -m compileall "path_to_your_sendmail.py_file"

6. 'Upload' and overwrite the file (sendmail.py) to MotionEyeOS by using WinSCP again (location: /usr/lib/python2.7/site-packages/motioneye)

7. Done, but remember that with an update of MotionEyeOS your changes can be lost.

closed 2017-02-23 by bortek