motioneye-project/motioneyeos #45: Offline mode

perardf, 2015-11-25:
Will it be possible to have an offline mode in order to detect motion even in place without internet access ? I would like to get that in order to install it on construction site.

thanks in advance for your help

ccrisan, 2015-11-25:
motionEyeOS is not designed to be ran without internet connection. You can try it if you want but you will definitely won't have the correct system time.

closed 2015-11-25 by ccrisan

ektus, 2015-12-11:
Without correct system time, it won't record any images.With link watchdog active, the system will reboot upon losing the link, thus will also reset the time to 1970 and cease operation. Not a good behaviour for a security device, IMHO.

Regards
Ektus.

ccrisan, 2015-12-12:
I don't see the connection between the *link* and the system time, but I agree that the general behaviour when dealing with current date/time can and should be improved.

ektus, 2015-12-12:
By default, the link watchdog is active. For my first tests, I installed the system and let it running, sitting on the window sill. Then went off to work, switching off the router and all other equipment except the RPi. Result: Reboot of the RPi by the watchdog, failure to get time and date and IP-address due to unavailability of the connection, and no recording whatsoever, even with date 1970. Disabling the watchdog the next day did allow the RPi to stay up and running albeit the router being off again. So the offline condition doesn't necessarily stop the motioneyeos system, but it won't recover from any reboot during that time (power outage, whatever). I haven't been investigating the exact cause, though.