motioneye-project/motioneyeos #46: Intermittently web server stops responding

mneumark, 2015-11-27:
I set up a couple USB cameras on my Raspberry Pi 2, and then went out of town. It worked great for a day or so. Then out of nowhere the website stopped responding.

I can still get to the raw MJPEG streams, but the website no longer functions and motion is not running either (I can tell because I stopped getting notifications).

I cannot fix anything now because I'm out of town and did not set up SSH access (mainly because the web UI does not allow you to customize the SSH port).

When I get back into town, is there a particular log that will help you troubleshoot this?

ccrisan, 2015-11-27:
1) If you can see the MJPEG stream, then motion is definitely running.
2) SSH is enabled by default and you should never connect motionEye directly to the Internet. Instead, you should place it behind your router and selectively forward the required ports (preferably not 80 and 22 directly).

What usually happens when the port 80 on motionEyeOS is directly accessible from the Internet is that you get scanned, and the motionEye web server gets overwhelmed with unwanted HTTP requests. Thus it becomes nonresponsive.

The only interesting log file in your case (assuming that you have indeed connected your unit directly to the Internet and I'm not mistaking) is motioneye.log, which probably contains the various unwanted HTTP requests.

mneumark, 2015-11-27:
Since I last posted, my MJPEG feed no longer works either.

I have the web port opened on my router on a random non-80 port (48255), the two feeds are at 48256 and 48257.

I just tried tying 48258 on the outside to 22 on the inside to my network to the Pi (I can still access my router's config). I cannot seem to SSH into it. Seems I can't access it in any way now.....guess I'm SOL for now. I'll let you know in a couple days when I can access the motioneye.log file.

ccrisan, 2015-11-27:
In that case I'd be interested in all the log files before having an idea of what has been going on.

mneumark, 2015-12-01:
@ccrisan Here is a copy of all the log files as requested:
https://gist.github.com/mneumark/2b608550e08f22e17c2f

ccrisan, 2015-12-02:
1) Are you using the Fast Network Camera option?
2) If not, are you using a SMB (network) share as your root directory?

I'm seeing different details in your log files and I couldn't tell what your configuration is. Could you please describe how you configured your motionEyeOS and how exactly you're using it?

mneumark, 2015-12-02:
1) I did experiment with the Fast Network Camera option but decided I did not like the lack of notifications and so I disabled it.
2) I tried setting that up but didn't have time to figure out what I was doing wrong so I gave up for the moment.

I have 2 cameras hooked in. One is the Raspberry Pi Camera and the other is a Logitech 1080p web cam. I was primarily logging in remotely to the web to check on my dog. I had the resolution of both set to 720p. For both, I had it set to make a curl command to trigger a PushOver notification when there was movement (my dog mostly just sat there so I wanted to know when she moved or when someone else came in).

It was just recording locally onto the MicroSD card. I DID experiment with a few things but that is how I finally set it.

Does that help?

ccrisan, 2015-12-02:
Issues with Logitech cameras are known and people have reported various crashes and malfunctions when using various Logitech USB camera devices. Lowering the resolution seems to help in many cases.

melashkov, 2016-02-07:
I have been using motioneyeos for over a year and started experiencing same issue recently, I thought it is due to an update problem, config compatibility, so I had a fresh install of the motioneyeos and after configuring wifi, rasperry pi is not responding over http, ssh, or samba but motionpi is still running. @ccrisan would you please tell me what is the best way to debug this issue. thank you, and yes I am using logitech camera :smile:

ccrisan, 2016-02-07:
Attaching your log files could be a good start.

melashkov, 2016-02-07:
@ccrisan where are the logs?

ccrisan, 2016-02-07:
They are under the Expert Settings settings menu section. If UI is no longer accessible, you can find them in /data/log, or in the /log folder on the third partition of your SD card.

bortek, 2017-02-23:
I consider this is no longer an issue therefore closing.

closed 2017-02-23 by bortek