motioneye-project/motioneyeos #47: web update

funtomas, 2015-11-29:
Software update can't find new version 
"motionEye is up to date (current version: 20151103)"

scoobyd00, 2015-11-29:
It's probably because it's a pre release and not the full update.
Think there's an option to select beta releases if you want to  try it

funtomas, 2015-11-29:
Ithanks i found "PRERELEASES" option

funtomas, 2015-11-29:
in FF update proces failed 
then i have used IE and it was successful

ccrisan, 2015-11-30:
I'd like to see the logs.

tokodude, 2015-12-02:
Mine won't update ether.
I put on the Pre-Release option and rebooted, but it still say's I have the latest version installed.

ccrisan, 2015-12-02:
What version do you currently run @tokodude?

tokodude, 2015-12-02:
Current Version	20151103

ccrisan, 2015-12-02:
Could you please also enable debugging, reboot, try to update again and, if it still says it's up to date, please send me the motioneye.log file.

tokodude, 2015-12-02:
Here:

[motioneye.txt](https://github.com/ccrisan/motioneyeos/files/49736/motioneye.txt)

ccrisan, 2015-12-02:
Your board doesn't seem to have access to the internet (at least to to a DNS server that works):

    1970-01-01 01:00:49: [motioneye]    DEBUG: fetching https://api.github.com/repos/ccrisan/motioneyeos/releases?_=49...
    1970-01-01 01:01:09: [motioneye]    ERROR: could not get versions: <urlopen error [Errno -3] Temporary failure in name resolution>

closed 2015-12-02 by ccrisan