motioneye-project/motioneyeos #49: Support for SSL?

chaseacton, 2015-12-01:
Does motionEyeOS have support for SSL? When logging in remotely, my credentials are sent in plain text.

ccrisan, 2015-12-01:
Nope. SSL (HTTPS) is not the job of motionEye(OS), but rather of a reverse proxy (e.g. nginx) placed in front of it. HTTPS comes together with a certificate, which is attached to a domain.

Your credentials are never transmitted in fact. motionEye uses a signature mechanism that prevents (at least) the leak of your credentials.

closed 2015-12-01 by ccrisan

jjanderson, 2021-02-02:
Hello,

I run multiple services using my reverse proxy application in synology. I however just cannot get motion eye to connect using this method. Apologies fopr raising this thread again, but I have literally tried everything... 

I am running motioneye in docker on the synology box.

ItsANoBrainer, 2023-08-25:
Running nginx reverse proxy, and the page just constantly refreshes when setup properly for SSL and HTTPS.