motioneye-project/motioneyeos #50: Multi cams, Philips pc 900 crash after few minutes

L3o-pold, 2015-12-02:
I have a Raspberry PI 2 model B with a Raspberry Pi camera module (black) and a Philips pc 900 (USB).
Philips pc 900 crash after 30 minutes and seems busy when motioneyeos tries to restart it.

Here is my log files
http://pastebin.com/kkrLxbuZ (motion.log)
http://pastebin.com/GsWzR248 (motioneye.log)
http://pastebin.com/DSLU8kMX (messages.log)
http://pastebin.com/eWEEkiuW (dmesg.log)

The interesting parts seems to be:

The crash log I presume.
```
[2] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[2] [NTC] [VID] vid_close: Closing video device /dev/v4l/by-id/usb-0471_0329_01690000CE000000-video-index0
```

And the unsuccessful automatique restart.
```
[2] [NTC] [VID] v4l2_do_set_pix_format: Testing palette YU12 (320x240)
[2] [ERR] [VID] v4l2_do_set_pix_format: Error setting pixel format.
VIDIOC_S_FMT: : Device or resource busy
[2] [ERR] [VID] VIDIOC_TRY_FMT failed for format v4l2_set_pix_format: Device or resource busy
[2] [ERR] [VID] v4l2_set_pix_format: Unable to find a compatible palette format.
[2] [NTC] [VID] vid_v4lx_start: Using V4L1
[2] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
```

ccrisan, 2015-12-02:
It seems like a buggy driver or something fishy at a lower level. I wouldn't know what to do about this. Does it work on Raspbian with motion?

L3o-pold, 2015-12-02:
Yes it works but I will try with a longer test session. The camera works out of the box by the way.

L3o-pold, 2015-12-06:
Looks like activating Automatic Brightness for the Phillips cam fix the issue.
It think that the cam timeout after some inactivity and the automatic brightness option keeps it alive.
I noticed that when the cam crash, you can restart the stream by pressing the top button (the button is supposed to take a picture), and you don't need to restart the Rasp.

closed 2015-12-06 by L3o-pold