motioneye-project/motioneyeos #52: The list of days are out of focus!

breakthecode, 2015-12-02:
Hi!
Using OS X El Capitan on my iMac 27" with 5K res. with latest Firefox and as you can see, I can only see todays date, and just above that I can see the beginning of the last days box. What could be the issue?

[image]

Best
Pontus

ccrisan, 2015-12-02:
What version of motionEyeOS are you using?

breakthecode, 2015-12-05:
I´m using the 20151103 running on my Raspberry Pi 2 model B.

ccrisan, 2015-12-06:
The issue has been fixed in 20151129, which is a prerelease. Please enable pre-releases to be able to upgrade.

closed 2015-12-06 by ccrisan