motioneye-project/motioneyeos #53: Motion Detection  - Run a Command

fdarrigo14221, 2015-12-04:
Hello.  I would like MotionEyeOS to run a python script whenever it detects motion.  The Python command creates a log file which I am importing into my SIEM.  I confirmed the Python script is error-free but I can't tell if the script is being called whenever motion is detected.  My command line is 

python sdcard/motionlog/motionlog.py

On another note, I would like the log to be saved on a windows share, but whenever I run the motionlog.py via SSH, the file is created in the sdcard/motionlog directory.

Advice is greatly appreciated

ccrisan, 2015-12-04:
Saving the log file on a network share could be difficult and I wouldn't be able to help you with that. motionEyeOS only supports writing log files locally, on the data partition of your SD card.

closed 2015-12-04 by ccrisan

fdarrigo14221, 2015-12-04:
Thanks for the reply.  I understand.  I elected to configure the email notification and write the email to a text file which can be imported into my SIEM.