motioneye-project/motioneyeos #55: foscam fi8904w not a supported network camera

zorrotenai, 2015-12-04:
Hi all; I bought a foscam fi8904w because I read it's a motion compatible device, but I get "not a supported network camera"
[image]

psts, 2015-12-05:
Try this url instead: rtsp://192.168.1.101:88/videoMain

Taken from: http://www.ispyconnect.com/man.aspx?n=foscam

zorrotenai, 2015-12-05:
Thanks first off all...
a little step: now the problem is "authentication failed"
 I reseted from factory my camera, so I'm sure user is admin whit no password, but nothing: 
[image]

zorrotenai, 2015-12-05:
OK, this work:  http://<ip>:<port>/videostream.cgi?user=<user>&pwd=<password> from https://bitbucket.org/ccrisan/motioneye/issues/190/foscam-camera-is-not-working
Thank you very much

zorrotenai, 2015-12-05:
I found this working:  http://<ip>:<port>/videostream.cgi?user=<user>&pwd=<password> from https://bitbucket.org/ccrisan/motioneye/issues/190/foscam-camera-is-not-working
Thank you very much

closed 2015-12-06 by ccrisan