motioneye-project/motioneyeos #56: [sendmail] ERROR: failed to send mail: Connection unexpectedly closed

breakthecode, 2015-12-05:
Hi again!
Have checked all that I could, but my logs are full of this.

2015-12-05 13:58:09: [sendmail]    ERROR: failed to send mail: Connection unexpectedly closed
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 194, in on_message
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 51, in send_mail
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
  File "/usr/lib/python2.7/smtplib.py", line 317, in connect
  File "/usr/lib/python2.7/smtplib.py", line 368, in getreply
SMTPServerDisconnected: Connection unexpectedly closed
2015-12-05 13:58:47: [sendmail]    ERROR: failed to send mail: Connection unexpectedly closed
Traceback (most recent call last):
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 194, in on_message
  File "/usr/lib/python2.7/site-packages/motioneye/sendmail.py", line 51, in send_mail
  File "/usr/lib/python2.7/smtplib.py", line 256, in __init__
  File "/usr/lib/python2.7/smtplib.py", line 317, in connect
  File "/usr/lib/python2.7/smtplib.py", line 368, in getreply
SMTPServerDisconnected: Connection unexpectedly closed

The username and PW is correct and working for Gmail.

breakthecode, 2015-12-05:
Update: That´s because I was using 465 as the port. When I changed it to 587 Gmail refuse to let me use it due to insecure app.

breakthecode, 2015-12-05:
Update 2: For the other ones that keeps getting this errors: https://support.google.com/accounts/answer/6010255?hl=en
Maybe a note around this in the Wiki?

L3o-pold, 2015-12-06:
I don't have any issue with Gmail account and TLS option activated on 587 port.

donnlee, 2016-12-02:
I have a lot of people moving around the house and that easily exceeds the max number of messages that Gmail allows me to send per day.   That was a problem for average days, but not for vacations / holiday.
Solution: I configured my VPS as a private Postfix SMTP server, so now I can send as many emails as I want.

closed 2017-02-23 by bortek