motioneye-project/motioneyeos #57: sound recording?

shinji2009, 2015-12-06:
hi. is it possible to add audio sources? like usb sound card or some board have mic in. and assosiate it with video sources so record video with sound?

ccrisan, 2015-12-06:
No.

shinji2009, 2015-12-06:
thanks for answer. thats a bummer ( it will be realy useful ((

mitchelldavis44, 2016-03-07:
+1 on the sound issue.

Mugiwara75, 2016-04-30:
+1 for sound ;-)

Mayday171, 2016-06-06:
+1 mic support would be great

spyrotsk, 2016-07-23:
+1 for sound support!
Thanks! ;-)

hellyberry, 2017-01-07:
Sound and universal sensor action support would be great, for monitoring PIR, Temperature, Moisture, ...

bortek, 2017-02-23:
This is duplicate of #57

closed 2017-02-23 by bortek

shinji2009, 2017-02-23:
it is #742 is duplicate of this request, it is 25 days old, mine is more than a year

bortek, 2017-02-23:
Does not matter really which Issue we will be driving since it is the same feature request. Lets keep #742 ac active and keep this one closed. Thanks!