motioneye-project/motioneyeos #58: How to reach the cam by internet?

zorrotenai, 2015-12-06:
I tried to portforward 
[image] but I can't reach the cam 
[image]

ccrisan, 2015-12-06:
It would be useful to tell us a bit more about your network layout, and of course what exactly you are trying to achieve.

As far as I can tell, you have at least three devices involved: a motionEyeOS-based board and two Foscam network cameras. Are they on the same LAN? Why do you want to set up port forwarding?

zorrotenai, 2015-12-07:
Thanks, Calin, you are an angel .. more Archangel ;-)
I'm newby; whith foscam setup was easy to reach the camera by internet, but I want it by motioneyeos, so I tried to portforward, but I don't know well what I'm doing 
I want to install at the moment only one camera in raspberry + motioneyeos in my lan and to see the streaming by internet.
You saw 2 devices because I have not assigned yet a fixed address and the router change it, so I tried with different port to open
My lan is only modem router + pc + smartphone + raspberry + cam

ccrisan, 2015-12-08:
If your camera is inside LAN, then add it using its private IP address. It will be available in motionEye after forwarding to port 80.

closed 2017-02-23 by bortek