motioneye-project/motioneyeos #59: Brownout

Abrastreve, 2015-12-07:
After installing the new 20151206 from download (update of 20151103 failed) I keep on getting the message: <brownout: low power supply voltage detected> at regular intervals on the monitor attached to my RaspberryPi-2. Nevertheless, MotionEyeOS does work despite this message.
Switching power supplies does not help and the message does not appear when I run other applications on this machine or when I revert back to 20151103.

ccrisan, 2015-12-08:
Brownout has been added to recent versions. Normally you could ignore it if everything works fine, but I suggest making sure the power supply is good enough for your setup. The warning appears when the voltage drops below 4.75V. This can be caused by a PSU that cannot supply enough current, by using long power wires or by adding USB devices that draw too much current.

Abrastreve, 2015-12-10:
Thanks for your answer Calin.
In the meantime I have played around with 20151206 on 2 different RPi's and 4 different power supplies plus one wireless keyboard dongle and one camera in the USB ports.
The outcome was that the RPi model-1 did not give any brownout low-power messages for any of the 4 tested power supplies and that the PR model-2 gives a brownout warning for all power sources.
Anyway, I'll keep using the RPi-2 despite the brownout warning as it does function well.
Thank you for a wonderful piece of software and issue closed as far as I am concerned.

closed 2015-12-10 by Abrastreve

ccrisan, 2015-12-12:
@Abrastreve the PI B (not B+) is doesn't know how to provide the brownout information to the OS, that's why you don't see it.

I would however like to find out what causes the brownout with your PI2, to make sure it's not a false positive.

1) What current rating do your power adapters have? Are they all 5V?
2) What USB devices have you connected to your PI2? Have you tried it w/o (some of) those devices?
3) Have you by any chance connected something on one of your GPIOs?
4) Do you by any chance have a voltmeter/multimeter at your disposal so you can measure the actual voltage level at the pins of your PI?
5) Would you be able to set up remote access for me to log in and see if it's a problem with the detection mechanism itself?

Abrastreve, 2015-12-14:
Hello Calin,

Following are the answers to your five questions:
1) Current ratings of my 4 power units are respectively: 2.4 A ; 2.1 A ; 1
A and 1 A
2) Connected to the Pi-2 are a USB wireless keyboard dongle and a PiNoIR in
the camera socket. I have tried it without these but that makes no
difference.
3) There is nothing connected to any of the GPIO pins.
4) I have connected two multimeters to the power cable to measure current
and voltage. The voltage ranges from 4.98 V to 5.07 V among the four power
supplies while current remains fairly stable at around 0.58 A and sinks to
0.44 A when USB dongle and camera are removed. Two pics of my measurement
setup have been attached to this mail.
5) Remote access should be possible. You need an IP address, portnumber,
login name and a password from me I suspect? To where can I send these
safely?

ccrisan, 2015-12-15:
1, 2, 3, 4) Then I believe you're seeing false positives. Let's try to see why.
5) Yes, please email them to me at ccrisan at gmail dot com.

tuxedo0801, 2016-08-31:
Any update on this?

I also have a 2,4Amps power supply on raspberry pi 2. Power-Cord is about 1m. Beside power-cord, there's only ethernet + raspicam connected. No USB device.

And I'm also getting this messages with latest release :-(