motioneye-project/motioneyeos #60: Feature

psts, 2015-12-10:
Not an issue, but a feature request. Let me explain my setup first.

I have several Pi2's connected to my outdoor cameras and then one Pi1 connecting them all, so I only have one interface to look at from the phone or desktop browser.

Each camera's motion detection goes to a shell script that sends a push notification thru Pushover.

The issue is if I want to disable motion detection, I have go to each camera and disable "Run a command". When I enable it again, I have to write everything back again under "Run a command", since it doesn't save the previous value.

So...

It would be great if we could make custom buttons that can do a few things. Then buttons should be easy to find in the main interface.

1. echo "enable" > /data/etc/buttons/motion_detection
Then you can check in your script if "motion_detection" is enabled or disabled.

2. Enable/disable GPIO pins

... to just name a few.

ccrisan, 2015-12-11:
Custom action buttons are planned. In fact they are already implemented in an experimental branch. I'm hoping to merge them into master within a month or so.

psts, 2015-12-11:
Looking forward to try it out!

ozett, 2015-12-28:
looking forward, too

ccrisan, 2015-12-30:
Custom buttons will be available with the soon-to-be-released build. They have already been commited to the master branch (of motionEye as well as of motionEyeOS). Do hold your breath :)

closed 2015-12-30 by ccrisan