motioneye-project/motioneyeos #61: Use local rtc with RPi 2 in case of network unavailability: hwclock command missing

ektus, 2015-12-11:
I'd like to use a hardware clock (DS3231, http://www.raspberrypi-spy.co.uk/2015/05/adding-a-ds3231-real-time-clock-to-the-raspberry-pi/) in order to set the time if no time server is available at boot (link down). The i2c modules are there, I was able to detect the clock and define the interface in userinit.sh, but can't actually set the clock because the hwclock command is missing. Could you add it the next revision, or can I copy it from a full Raspian install?

Regards
Ektus.

ccrisan, 2015-12-12:
Yes, hwclock is coming soon.

ektus, 2015-12-12:
great. I'm looking forward to it.

ccrisan, 2016-01-03:
I remember adding the command to the image. However it somehow still isn't there. Let's hope it'll make into the stable release :)

GubblGubbl, 2016-01-04:
Hi ektus,

If you need a fast solution even before the next release, I could offer you a script I wrote for the exact same purpose. It reads the I2C bus information from a RTC and saves it to the system clock.

ektus, 2016-01-05:
Thanks, but as I've since switched to Raspbian Wheezy, I don't need it. Others might, though, so why not post it here?

milo3, 2016-02-10:
Hi GubblGubbl,
could you please share your script? I am facing the same issue.

GubblGubbl, 2016-02-10:
Sure, I'll post it in the evening (UTC-1)

GubblGubbl, 2016-02-10:
I have not forgotten you, I have to search the script's location.

milo3, 2016-02-11:
Hello,
I've created `/data/etc/userinit.sh` file including these lines:

`modprobe rtc-ds1307`
`echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-1/new_device`
`hwclock -s`
`date`

It seems to work. Thank you anyway!

alinmiron, 2016-07-20:
hi guys, any news about support for hardware clock (DS3231/DS1307) out of the box? or should I use a similar script as the one posted by @milo3.
Thank You

GubblGubbl, 2016-07-20:
The way milo3 wrote works well. Just use the right RTC modell in the commands.

anjoze, 2016-10-31:
Tested with Rpi2 + version 20161027 + DS3231
Not working :(

ccrisan, 2016-12-04:
@anjoze the hwclock command is there, the modules are there (apparently yours is called `rtc-ds1307`, as [this](http://www.raspberrypi-spy.co.uk/2015/05/adding-a-ds3231-real-time-clock-to-the-raspberry-pi/) article says). What exactly doesn't work for you? What errors do you get?

milo3, 2016-12-04:
As I wrote on #549,  I don't see i2c-1 on /sys/class/i2c-adapter/

koenig-dro, 2016-12-09:
Same problem to mine...tried some things...
is there anybody who describe from out of the box to running ds3231?

ccrisan, 2016-12-09:
It appears that the i2c hardware interface needs to be enabled by adding `dtparam=i2c_arm=on` to `/boot/config.txt`. The next version will have this enabled, as well as the i2s, spi and audio interface.

closed 2016-12-09 by ccrisan

anjoze, 2016-12-15:
Ok, finally it's working :) but I had to do some changes.
I will describe step by step:
**Image used: 20161212**

**1.** Enable writing files (https://github.com/ccrisan/motioneyeos/wiki/FAQ#why-do-i-get-read-only-file-system-when-trying-to-edit-a-file)
```
mount -o remount,rw /
mount -o remount,rw /boot
```

**2.** `nano /etc/modules`
Add this at the end: **rtc-ds1307**
   (Maybe you need to reboot the system... I don't remember)

**3.** Execute this : `echo ds1307 0x68 > /sys/class/i2c-adapter/i2c-1/new_device`

**4.** That is all. Now you can execut: `hwclock`
It will show the RTC date
Than you can easily set the date from the internet:
`ntpd -gq`
`hwclock -w`

Thanks again ccrisan for all the great work.

ccrisan, 2016-12-16:
Thanks for putting together all these instructions. I wrote a small [tweaking article](https://github.com/ccrisan/motioneyeos/wiki/Tweaks#adding-a-rtc-module) on this matter just a few days ago. It's similar to what you have done, but uses the `/data/etc/modules` file which is persisted across updates and does not require remounting the system partition in read-write mode.

thepiotrus, 2017-04-20:
A solution for RTC module ds3231 tested with Raspi 3 and MotionEye version 20170212 
(Check MotionEye release with **Adding an i2c RTC module** , 20161211 and up)

`mount -o remount,rw /` 
`mount -o remount,rw /boot`
After create or edit modules with the command  (becareful don't forget  **/data**/etc...)
`nano /data/etc/modules`  

And add this line
`rtc-ds3231`

After create or edit userinit.sh with the command (becareful **/data**/etc...)
`nano /data/etc/userinit.sh`

And add this 3 lines
```
echo ds3231 0x68 > /sys/class/i2c-adapter/i2c-1/new_device
sleep 5
hwclock –s
```
**sleep 5** for avoids the error : can't open /dev/misc/rtc : no such file or directory failed

`Reboot` 

If you don't have internet connexion change date manualy 
`date MMDDHHMMYY`

For finish write date to the module ds3231 with the next command
`hwclock -w`

and `reboot` 

Thanks a lot ccrisan for your work !!!

ccrisan, 2017-04-21:
@thepiotrus I added the sleep command to the wiki article. Thanks.

PhotoKevin, 2017-05-08:
Dec. 16th ccrisan said:

> It appears that the i2c hardware interface needs to be enabled by adding dtparam=i2c_arm=on to /boot/config.txt. The next version will have this enabled, as well as the i2s, spi and audio interface.

I just upgraded to the 20170329 version and config.txt doesn't have this.

ccrisan, 2017-05-09:
@PhotoKevin, what board? What does your `/boot/config.txt` look like now?

PhotoKevin, 2017-05-09:
It's a Raspberry Pi 2 Model B.  To give you the /boot/config.txt will have to wait until I get home from work.  I went through the remounting described above and added the dtparam line in, and everything started working.

PhotoKevin, 2017-05-10:
@ccrisan My current config.txt is:

```
start_file=start.elf
fixup_file=fixup.elf
arm_freq=700
core_freq=250
sdram_freq=400
over_voltage=0
gpu_mem=128
gpu_mem_256=128
gpu_mem_512=128
disable_camera_led=1
gpu_freq=250
dtparam=i2c_arm=on
```

That last line is the one I added manually.

elfhelmp, 2017-07-11:
In order to make sure your ds3231 RTC or DS1307 actually works, make sure to check with command `i2cdetect -y 1` and it will show something like this:

[root@XXXXXXX ~]# i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- 68 -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

If another number is shown instead of 68, then you need to update file `/data/etc/userinit.sh` with the corresponding number.