motioneye-project/motioneyeos #62: Feature request: Support for custom resolutions

ektus, 2015-12-11:
I've got a 3MP IP camera (QCAM, http://www.amazon.de/gp/product/B014E995K8), that's got the following settings for the output stream:
2304*1296
2048*1536
1080p (1920x1080)
720p (1280x720)
Motioneyeos on a RPi 2 connects just fine and is able to detect motion as well as stream and save images, but none of the camera resolutions is available in the Motioneyeos camera settings. 

The streaming image constantly changes size by a couple pixels (every second or faster) and there are compression artefacts and thousands of false positives with the motion detection. I've got a 1TB USB harddisk attached, but the web interface becomes unresponsive with thousands of images.

I'd like to set one of the native camera resolutions in order to minimize re-sampling errors and probably steady the image, preferably the 3 mega pixel one because that's the only one with the full field of view. The camera stream is set to 6 frames per second with vbr and max. 2048kbps.

Is there a way to edit some configuration file to add more resolutions? The 720p and 1080p resolutions should be added to the default list, as well as a "custom" entry with manual setting of x and y resolutions.

Regards
Ektus.

ccrisan, 2015-12-12:
Indeed adding the ability to manually set the width and the height is a good thing to have.

Chiny91, 2015-12-20:
Well... having 3 identical cameras, I just added:
width 704
height 576
to /data/etc/motion.conf
*below* the **thread** lines, which means motion.conf takes priority.   I have found the GUI does not trample over these additional lines.   Yes, the image size does end up being width*height.

Of course, adding those lines to the individual thread (camera) files would work but I suspect subsequent use of the GUI would undo these manual modifications.

Note; motion wants the values in width/height to be a multiple of 16.

Careful thought around the position of the width/height lines, and the content of the thread files, should add a little more flexibility.

ccrisan, 2016-11-16:
Fixed with https://github.com/ccrisan/motioneye/commit/f1298c012312aee69e231d2831abab2c3d76a0a6 and https://github.com/ccrisan/motioneye/commit/d61858b0ad47a13428e850ee475619ac57e0af62.

closed 2016-11-16 by ccrisan

ektus, 2016-11-16:
Works as expected, thank you very much!