motioneye-project/motioneyeos #64: Issue connting to Hikvision and Huacam

SMFabbro, 2015-12-13:
Hello, I am trying to use a Hikvision and Huacam IP camera.  when I enter in the IP address I get the following error;
Recv failure: Connection reset by peer
- or -
not a supported network camera

Need some help please - does the software support these types of cameras?
under the URL line I am entering http://192.123.1.181:8081/ or http://192.123.1.183.80/

Thanks

Chiny91, 2015-12-14:
I have 3 Hikvision cameras working happily, all type HIKVISION DS-2CD2032-1.   I forget what the URLs you are entering might do (snapshots perhaps) but try:
http://192.123.1.181/Streaming/channels/1/httpPreview (set substream for mjpeg)
or
rtsp://192.123.1.181/h264/ch1/main/av_stream (I find that guzzles R Pi cpu power)

If that does not work, you may need to change the camera firmware - an awkward task.

SMFabbro, 2015-12-14:
Thanks...that worked for my two Hikvision....still trying to get the Huacam's to work

closed 2017-02-23 by bortek