motioneye-project/motioneyeos #66: Odd movie behaviour

rpiuser, 2015-12-14:
Anyone experiencing the recording video suddenly speeding up half the way through ? On raspberry pi with raspbian install - I've tried different speed settings and they all do the same

Gordon-1, 2016-02-25:
Yes, all mine do that.  I have not investigated further so I can say no more about it,

ccrisan, 2016-02-25:
Is this a RPI1 we're talking about? If yes, please try a RPI2. The processing power might not be sufficient.

Gordon-1, 2016-02-25:
In my case it is a RasPi 2, RasPi camera and LogiTech C210 USB cam. The first few frames play at normal speed but then it is like someone pressed a fast-forward button.  I am concentrating on getting good stills for now, since that is sufficient for security but I can look into this more when I have time.

BTW, I need to check my hardware and settings to see if I have a problem because I just got this in the motioneye.log:
2016-02-25 08:40:29: [motioneye]     INFO: server started
2016-02-25 09:18:15: [motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed
2016-02-25 09:18:15: [motioneye]  WARNING: mjpg client connection for camera 2 on port 8082 is closed
2016-02-25 09:18:42: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2016-02-25 09:18:48: [motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed
2016-02-25 09:18:48: [motioneye]  WARNING: mjpg client connection for camera 2 on port 8082 is closed
2016-02-25 09:19:56: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2016-02-25 09:20:01: [motioneye]    ERROR: mjpg client for camera 1 on port 8081 error: [Errno 104] Connection reset by peer
2016-02-25 09:20:01: [motioneye]    ERROR: mjpg client for camera 2 on port 8082 error: [Errno 104] Connection reset by peer
2016-02-25 09:20:01: [motioneye]  WARNING: mjpg client connection for camera 2 on port 8082 is closed
2016-02-25 09:20:01: [motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed
2016-02-25 09:20:01: [motioneye]    ERROR: connection problem detected for mjpg client for camera 2 on port 8082
2016-02-25 09:20:07: [motioneye]    ERROR: mjpg client for camera 2 on port 8082 error: [Errno 32] Broken pipe
2016-02-25 09:20:07: [motioneye]    ERROR: mjpg client for camera 1 on port 8081 error: [Errno 32] Broken pipe
2016-02-25 09:20:07: [motioneye]    ERROR: connection problem detected for mjpg client for camera 2 on port 8082
2016-02-25 09:20:12: [motioneye]    ERROR: connection problem detected for mjpg client for camera 1 on port 8081
2016-02-25 09:20:17: [motioneye]  WARNING: mjpg client connection for camera 1 on port 8081 is closed
2016-02-25 09:20:17: [motioneye]    ERROR: mjpg client for camera 2 on port 8082 error: [Errno 104] Connection reset by peer
2016-02-25 09:20:17: [motioneye]    ERROR: connection problem detected for mjpg client for camera 2 on port 8082
2016-02-25 09:20:22: [motioneye]    ERROR: Stream is closed
Traceback (most recent call last):
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/web.py", line 1369, in _stack_context_handle_exception
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/stack_context.py", line 314, in wrapped
  File "/usr/lib/python2.7/site-packages/motioneye/mjpgclient.py", line 118, in _on_connect
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/iostream.py", line 368, in write
  File "home/ccrisan/motioneyeos/output/raspberrypi2/target/usr/lib/python2.7/site-packages/tornado/iostream.py", line 878, in _check_closed
StreamClosedError: Stream is closed
2016-02-25 09:20:22: [motioneye]  WARNING: mjpg client connection for camera 2 on port 8082 is closed

ccrisan, 2016-02-25:
Try removing the USB camera as it seems to cause some troubles. motion.log and dmesg.log could contain additional details on this matter.

Gordon-1, 2016-02-25:
Will do.. although it seemed to be working well since I got the cam two days ago.
In the meantime can you interpret this line from motion.log:
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD

Also, dmsg output is attached (with USB cam still on).
[dmesg.txt](https://github.com/ccrisan/motioneyeos/files/146713/dmesg.txt)

Gordon-1, 2016-02-29:
Video is still as described using a Pi 2 and the Raspi camera only.  Let me know if you want to see a sample - I can email you a link to watch it.
Looks like I actually got an order in for the RasPi 3!  So should be able to try motioneyeos on that in a few days.

ccrisan, 2016-02-29:
motionEyeOS won't work with the RPI3 without modifications. Yes, please email me the link so I can see what this is about.

Gordon-1, 2016-03-01:
I am starting to suspect it is nothing more than a bottleneck due to the frame rate, resolution and quality settings. I was able to get video with no "stuttering" and apparently few or no dropped frames at lower settings but higher settings progressively produced poorer video. I wanted to blame my class 2 microSD card however saving to a USB flash drive made no difference.

I started with a clean install using ethernet to eliminate other variables such as my mask file.

320x240 2 fps 25% quality:  No apparent drops in video.
320x240 15 fps 85% q.: Still good but one gap
640x480 15 fps, 85 q: Lots of stuttering (both microSD and usb flash drive)
800x 480, 5 fps, 75 q,:  Good video but of course still only 5 fps. (both microSD and usb flash drive)
800x480, 10 fps, 85 q.:  Moderate amount of stuttering.

I'm still concentrating on getting good photos since that will be better to ID a person and that is working well.

bortek, 2017-02-23:
Have not hear any updated since over a year , closing this one. Reopen if it is still a problem.

closed 2017-02-23 by bortek