motioneye-project/motioneyeos #69: Troubleshooting Wifi help needed

68hc11, 2015-12-17:
So I've got 2 Model B & a Pi2B running but I can not get any of these to connect wirelessly.

I am using a Trendnet TEW-648UBM that runs well with my OSMC Pi setup.

The MotionEye logs seem to indicate that the adapter is in place but can't connect to my router.  (logs show Configuring wireless network: no device if adapter is not plugged and Configuring wireless network: no link if it fails to reach my router).

My SSID is set properly, case sensitive.

I am not using any security, hence no key (would this cause a problem)?

any suggestions?

partial log...

Configuring wireless network: no link
 * Setting current date using http: failed
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Setting smb admin password: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
---- shutting down motionEyeOS 20151206 ----
 * Stopping motioneye: done
 * Stopping smbd: done
 * Stopping nmbd: done
 * Stopping proftpd: done
 * Stopping sshd: done
 * Stopping crond: done
 * Stopping date updater: done
 * Stopping network: done
 * Stopping wpa_supplicant: done
 * Stopping eudev: done
 * Stopping watchdog: done
 * Stopping brownout detector: done
 * Stopping syslogd: done
---- booting motionEyeOS 20151206----
 * Setting hostname: done
 * Loading kernel modules: done
 * Starting syslogd: done
 * Starting brownout detector: done
 * Starting watchdog: done
 * Starting eudev: done
 * Mounting filesystems: done
 * Setting root password: done
 * Starting wpa_supplicant: failed
 * Configuring wireless network: no link
 * Setting current date using http: failed
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Setting smb admin password: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.199/24
 # Default gateway is 192.168.1.1
 # DNS server address is 192.168.1.1
---- shutting down motionEyeOS 20151206 ----
 * Stopping motioneye: done
 * Stopping smbd: done
 * Stopping nmbd: done
 * Stopping proftpd: done
 * Stopping sshd: done
 * Stopping crond: done
 * Stopping date updater: done
 * Stopping network: done
 * Stopping wpa_supplicant: done
 * Stopping eudev: done
 * Stopping watchdog: done
 * Stopping brownout detector: done
 * Stopping syslogd: done
---- booting motionEyeOS 20151206----
 * Setting hostname: done
 * Loading kernel modules: done
 * Starting syslogd: done
 * Starting brownout detector: done
 * Starting watchdog: done
 * Starting eudev: done
 * Mounting filesystems: done
 * Setting root password: done
 * Starting wpa_supplicant: no device
 * Configuring wireless network: no device
 * Setting current date using http: failed
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Setting smb admin password: done
 * Starting smbd: done
 * Starting nmbd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.199/24
 # Default gateway is 192.168.1.1
 # DNS server address is 192.168.1.1

ccrisan, 2015-12-17:
The absence of a network key might might cause the problem. Although I'll need to run some tests myself.

68hc11, 2015-12-17:
I am happy to confirm that this is exactly the issue, if I supply a WPA key & set up security on the wifi router, Motioneye connects like a champ, if I try to join an open network it doesn't

wolverine-forever, 2015-12-25:
Having more or less the same issue, however in my case the device is not recognized .. stuff like lsusb isn't possible? I boot with the USB device connected.
I used a Netgear A6100. If I use it under windows or my linux machine it works fine.
I entered the networkkey as well as the SSID.

ccrisan, 2015-12-25:
@wolverine-forever lsusb is possible and recommended. The only problem is that you'll have to identify the device yourself using the vendor/product number (the usb device database is not included with motionEyeOS). Please use lsusb both in motionEye and your standard Linux distro with the WiFi dongle plugged in, to be able to identify it easily. Then report back to me the exact model of the dongle so I can see what's up with it.

In the meantime you could upload your log files here so I can try to identify the issue by looking at them.

wolverine-forever, 2015-12-26:
The ID is 0846:9052  (Netgear) i also tried it with  a D-Link 07d1:3c0a

---- booting motionEyeOS 20151206----
 * Setting hostname: done
 * Loading kernel modules: done
 * Starting syslogd: done
 * Starting brownout detector: done
 * Starting watchdog: done
 * Starting eudev: done
 * Mounting filesystems: done
 * Setting root password: done
 * Starting wpa_supplicant: no device
 * Configuring wireless network: no device
 * setting static gateway to 192.168.1.1: failed
 * setting static DNS server to 8.8.8.8: done
 * Setting current date using http: failed
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting motioneye: done
---- booting motionEyeOS 20151206----
 * Setting hostname: done
 * Loading kernel modules: done
 * Starting syslogd: done
 * Starting brownout detector: done
 * Starting watchdog: done
 * Starting eudev: done
 * Mounting filesystems: done
 * Setting root password: done
 * Starting wpa_supplicant: no device
 * Configuring wireless network: no device
 * setting static gateway to 192.168.1.1: failed
 * setting static DNS server to 8.8.8.8: done
 * Setting current date using http: failed
 * Starting http date updater: done
 * Starting crond: done
 * Starting sshd: done
 * Starting proftpd: done
 * Starting motioneye: done
 # Interface eth0 has IP address 192.168.1.60/24

[Thu Jan  1 00:00:01 1970] usb 1-1.1: New USB device found, idVendor=0424, idProduct=ec00
[Thu Jan  1 00:00:01 1970] usb 1-1.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[Thu Jan  1 00:00:01 1970] smsc95xx v1.0.4
[Thu Jan  1 00:00:01 1970] bcm2708_spi bcm2708_spi.0: master is unqueued, this is deprecated
[Thu Jan  1 00:00:01 1970] bcm2708_spi bcm2708_spi.0: SPI Controller at 0x3f204000 (irq 80)
[Thu Jan  1 00:00:01 1970] smsc95xx 1-1.1:1.0 eth0: register 'smsc95xx' at usb-bcm2708_usb-1.1, smsc95xx USB 2.0 Ethernet, b8:27:eb:77:5e:98
[Thu Jan  1 00:00:01 1970] usb 1-1.4: new high-speed USB device number 4 using dwc_otg
[Thu Jan  1 00:00:01 1970] usb 1-1.4: New USB device found, idVendor=0846, idProduct=9052
[Thu Jan  1 00:00:01 1970] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Thu Jan  1 00:00:01 1970] usb 1-1.4: Product: 802.11ac WLAN Adapter 
[Thu Jan  1 00:00:01 1970] usb 1-1.4: Manufacturer: Realtek 
[Thu Jan  1 00:00:01 1970] usb 1-1.4: SerialNumber: 00e04c000001
[Thu Jan  1 00:00:01 1970] EXT4-fs (mmcblk0p3): recovery complete
[Thu Jan  1 00:00:01 1970] EXT4-fs (mmcblk0p3): mounted filesystem with ordered data mode. Opts: (null)
[Thu Jan  1 00:00:07 1970] smsc95xx 1-1.1:1.0 eth0: hardware isn't capable of remote wakeup
[Thu Jan  1 00:00:09 1970] smsc95xx 1-1.1:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xDDE1
[Thu Jan  1 00:00:10 1970] random: nonblocking pool is initialized

This was done with ethernet cable in and dongle ... otherwise no access

1977475952] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[1977475952] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[1977475952] [NTC] [ALL] motion_startup: Motion mmaltest Started
[1977475952] [NTC] [ALL] motion_startup: Logging to syslog
[1977475952] [NTC] [ALL] motion_startup: Using default log type (ALL)
[1977475952] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD

I use the standard Raspberry Pi Cam module

Hope this info enough for you ...

ccrisan, 2015-12-26:
It seems as though the driver required for your device (rtl8812au) hasn't yet been included in the Raspberry PI kernel (which motionEyeOS uses). [This article](http://perversecontraptions.com/2015/06/06/how-to-get-a-netgear-a6100-wifi-usb-adapter-to-work-on-a-raspberry-pi/) explains how to add it yourself to a Raspbian installation. I won't maintain out-of-tree kernel drivers for motionEyeOS as it's too difficult, so I will mark this issue as "won't fix". Let's hope that the RPi kernel folks will soon consider integrating this driver themselves soon.

ccrisan, 2015-12-26:
Actually on a second read I realized the issue is about passwordless wifi networks which is in the works and will soon be available with motionEyeOS.

ccrisan, 2015-12-27:
The latest commit added support for open wifi networks.

joeybronner, 2018-07-17:
Hi @ccrisan, 

Is this a good example of passwordless wpa_supplicant.conf file? You see something wrong?
Because the configuration is not working and the OS is looping in the reboot mode.

Maybe for the first boot, it's required to use a regular WiFi with a password?

```
country=fr
update_config=1
ctrl_interface=/var/run/wpa_supplicant

network = {
    scan_ssid=1
    ssid="My-Awesome-Network"
    key_mgmt=NONE
}
```

Thank you for your help
</joey>