motioneye-project/motioneyeos #70: Enhancement request

scoobyd00, 2015-12-17:
Motioneyeos is running very well on my Pi 2. Very stable and excellent software.

One useful enhancement is relating to the motion detection and working schedule.

It would be really useful if we could have x2 time zones per day.

Ie for motion between 0900-1200 and 1800-2200 per day.

Thanks,

68hc11, 2015-12-18:
I encountered this as well, would be nice to span days...say start Friday night until Monday morning or even trigger on a GPIO input (if alarm is armed then Motioneye is also working).

ccrisan, 2015-12-18:
This is something that has been requested a few times. I'll take it into consideration, although this is definitely not a priority.

CampoES, 2016-12-23:
Yes, I'm too still interested in this feature.

Perhaps it is easier to manage timewindows like Monday 11:20 pm till Thuesday 09:00 am?

CampoES, 2016-12-23:
OK, you can set "outside working schedule" and set the time to for example 9:00 am to 04:00 pm. So the system should check for movement from 0:00 am to 8:59 am and 4:01 pm until 11:59 pm. That helps me.

bortek, 2017-06-01:
Closing this issue due to interest. If still interested please post your question in the forum page.

huvarda, 2017-08-23:
I would love to see this feature too.  I also want too have multiple working schedules.

jasaw, 2017-08-23:
An interface like this would be quite intuitive.
[image]

GSVNoFixedabode, 2018-10-10:
Would love to see this implemented! Any reason why this enhancement would be limited to motioneye rather than motioneyeos - too great a dependency on modules excluded from the *os version?
Cheers.