motioneye-project/motioneyeos #71: Time update

martymoose, 2015-12-18:
My time newer seems to update
always 1970
rpi2  latest version wifi
any ideas  what log i can check
is there a way to delay start until have network time
cheers  great os

ccrisan, 2015-12-18:
Do you have a direct (i.e. no proxy), working internet connection on your PI? Is your DNS properly configured? Please send me your `boot.log`, `dmesg.log` and `messages.log` files.

No there's no way to delay the startup process until time is available, simply because there are users that don't care about the current time.

martymoose, 2015-12-18:
hi thanks here you go
just needed the time to embed in video

*Martin Thompson*

*Po Box 1415*
*AYR   QLD    4807*

*M. 0466 005866*
*E.  marty@martymandy.com
<https://mail.google.com/mail/u/0/?view=cm&fs=1&tf=1&to=marty@martymandy.com>*
*W. **martymandy.com* <http://martymandy.com>

Check the weather at my place martymandy.com

ccrisan, 2015-12-18:
I'm sorry what video? Could you please send me the log files at ccrisan at gmail dot com?

MarekITPro, 2016-01-02:
My 3 pence: If Date Method is set to NTP it works fine and picks up the time, when set to HTTP it does not.

martymoose, 2016-01-02:
Hi mate just got home
i updated my 3 rpi to the new version
and they all work fine date from start works and embeded in stream fine saves in data folder as wel

thanks again

closed 2016-01-02 by martymoose