motioneye-project/motioneyeos #72: My setup is very unstable

digitizing, 2015-12-19:
First off this is the best user friendly software I have seen so far and looks very professional and want to thank you for both your efforts and support.  I am very motivated to figure out why I am have a few key issues.

MY SETUP: I have motionEye version 20151206 installed on Raspberry Pi 2 with a 8gig SD.  I have four Microsoft Lifecam HD3000 USB webcams all connected into an external powered USB hub.  For now I have most settings in default mode to reduce the troubleshooting variables.  I also have the Motion Detection turned off.  All four web cams are setup identically with 320x240 resolution.  I am hard wired to the network and have replaced both LAN cable and LAN switch to see if it is a network issue but am still having issues.  When I issue the TOP command via SSH access I see that the CPU is around 8%.  Is there any limitation on the number of locally connected cameras?  I do not see any information on typical setup scenarios and what are the limitations.  My application is for 4 to 6 cameras at home.  I plan on having two Raspberrry Pi 2s at different locations and hopefully both using the motionEye software. Right now I have my setup all on a table so it is easy to test and make changes.  I have also tried a variety of other tweaks earlier but no success. 

ISSUE 1: The issue is one of the four cameras is not streaming and is just a gray window.  When I first added the camera all was working.  Here is a portion of the motion.log that appears pertinent to my problem,

using motion binary /usr/bin/motion-mrdave
[0] [NTC] [ALL] conf_load: Processing thread 0 - config file /data/etc/motion.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-1.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-3.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-4.conf
[0] [NTC] [ALL] config_thread: Processing config file thread-5.conf
[0] [NTC] [ALL] motion_startup: Motion Unofficial-Git-UNKNOWN Started
[0] [NTC] [ALL] motion_startup: Logging to syslog
[0] [NTC] [ALL] motion_startup: Using default log type (ALL)
[0] [NTC] [ALL] motion_startup: Using log type (ALL) log level (WRN)
[2] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[1] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[3] [WRN] [VID] v4l2_select_input: Device doesn't support VIDIOC_G_STD
[4] [ERR] [VID] v4l2_select_input: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
[4] [ERR] [ALL] motion_init: Error capturing first image
[4] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device
[4] [WRN] [ALL] motion_loop: Retrying until successful connection with camera
[4] [ERR] [VID] v4l2_select_input: Error selecting input 0 VIDIOC_S_INPUT: Device or resource busy
[4] [ERR] [ALL] motion_loop: Video device fatal error - Closing video device

ISSUE 2: The other issue is I am getting quite a few random reboots. Here is a portion of the motioneye.log,

2015-12-19 08:18:20: [motioneye]  WARNING: Connect error on fd 18: ECONNREFUSED
2015-12-19 08:18:21: [motioneye]  WARNING: Connect error on fd 19: ECONNREFUSED
2015-12-19 08:18:22: [motioneye]  WARNING: Connect error on fd 19: ECONNREFUSED
2015-12-19 08:18:23: [motioneye]  WARNING: Connect error on fd 19: ECONNREFUSED
2015-12-19 08:18:24: [motioneye]  WARNING: Connect error on fd 19: ECONNREFUSED
2015-12-19 08:18:25: [motioneye]  WARNING: Connect error on fd 19: ECONNREFUSED
2015-12-19 08:18:26: [motioneye]  WARNING: Connect error on fd 19: ECONNREFUSED
2015-12-19 08:18:27: [motioneye]  WARNING: Connect error on fd 19: ECONNREFUSED
2015-12-19 08:24:49: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2015-12-19 08:24:56: [motioneye]    ERROR: could not terminate the motion process
2015-12-19 08:24:56: [motioneye]     INFO: rebooting
2015-12-19 08:24:56: [motioneye]     INFO: executing "/sbin/reboot"
2015-12-19 08:24:57: [motioneye]     INFO: interrupt signal received, shutting down...
2015-12-19 08:24:57: [motioneye]     INFO: server stopped
2015-12-19 08:24:57: [motioneye]     INFO: tasks stopped
2015-12-19 08:25:18: [motioneye]     INFO: hello! this is motionEye server 20151206
2015-12-19 08:25:21: [motioneye]     INFO: cleanup started
2015-12-19 08:25:21: [motioneye]     INFO: wsswitch started
2015-12-19 08:25:21: [motioneye]     INFO: tasks started
2015-12-19 08:25:21: [motioneye]     INFO: mjpg client garbage collector started
2015-12-19 08:25:21: [motioneye]     INFO: smb mounts started
2015-12-19 08:25:21: [motioneye]     INFO: server started
2015-12-19 08:25:21: [motioneye]  WARNING: Connect error on fd 14: ECONNREFUSED
2015-12-19 08:25:22: [motioneye]  WARNING: Connect error on fd 20: ECONNREFUSED
2015-12-19 08:25:23: [motioneye]  WARNING: Connect error on fd 20: ECONNREFUSED
2015-12-19 08:25:24: [motioneye]  WARNING: Connect error on fd 20: ECONNREFUSED
2015-12-19 08:25:25: [motioneye]  WARNING: Connect error on fd 20: ECONNREFUSED
2015-12-19 08:25:26: [motioneye]  WARNING: Connect error on fd 20: ECONNREFUSED
2015-12-19 08:25:27: [motioneye]  WARNING: Connect error on fd 20: ECONNREFUSED
2015-12-19 08:25:28: [motioneye]  WARNING: Connect error on fd 20: ECONNREFUSED
2015-12-19 08:25:29: [motioneye]  WARNING: Connect error on fd 20: ECONNREFUSED
2015-12-19 11:12:12: [motioneye]    ERROR: mjpg client timed out receiving data for camera 1 on port 8081
2015-12-19 11:12:19: [motioneye]    ERROR: could not terminate the motion process
2015-12-19 11:12:19: [motioneye]     INFO: rebooting
2015-12-19 11:12:19: [motioneye]     INFO: executing "/sbin/reboot"
2015-12-19 11:12:21: [motioneye]     INFO: interrupt signal received, shutting down...
2015-12-19 11:12:21: [motioneye]     INFO: server stopped
2015-12-19 11:12:21: [motioneye]     INFO: tasks stopped
2015-12-19 11:12:42: [motioneye]     INFO: hello! this is motionEye server 20151206
2015-12-19 11:12:45: [motioneye]     INFO: cleanup started
2015-12-19 11:12:45: [motioneye]     INFO: wsswitch started
2015-12-19 11:12:45: [motioneye]     INFO: tasks started
2015-12-19 11:12:45: [motioneye]     INFO: mjpg client garbage collector started
2015-12-19 11:12:45: [motioneye]     INFO: smb mounts started
2015-12-19 11:12:45: [motioneye]     INFO: server started
2015-12-19 11:12:45: [motioneye]  WARNING: Connect error on fd 16: ECONNREFUSED
2015-12-19 11:12:46: [motioneye]  WARNING: Connect error on fd 18: ECONNREFUSED
2015-12-19 11:12:47: [motioneye]  WARNING: Connect error on fd 18: ECONNREFUSED
2015-12-19 11:12:48: [motioneye]  WARNING: Connect error on fd 18: ECONNREFUSED
2015-12-19 11:12:49: [motioneye]  WARNING: Connect error on fd 18: ECONNREFUSED
2015-12-19 11:12:50: [motioneye]  WARNING: Connect error on fd 18: ECONNREFUSED
2015-12-19 11:12:51: [motioneye]  WARNING: Connect error on fd 18: ECONNREFUSED
2015-12-19 11:12:52: [motioneye]  WARNING: Connect error on fd 18: ECONNREFUSED
2015-12-19 11:12:53: [motioneye]  WARNING: Connect error on fd 18: ECONNREFUSED

Any assistance with these issues would be appreciated.

ccrisan, 2015-12-20:
Hi,

ISSUE 1: Please send me your `motion.conf` as well as `thread-*.conf` files, *removing any sensitive data (e.g. passwords)*. You can upload those files here, directly (please do not paste them though!)

ISSUE 2: I'd need to see your `boot.log` and `messages.log` files.

digitizing, 2015-12-20:
Thanks for your response and please see the log and conf files you requested in the attached zip file.

The configuration has changed a bit in these files as I do my troubleshooting.  The changes are,

1) Last night I did a new image from scratch
2) I still have a USB powered hub with four cameras
3) Increased the frame rate from 2 to 10 to try something different
4) Turned off motion detection
5) This morning I noticed lots of rebooting so decided to remove one of the cameras.  So currently I have three cameras but a little later I noticed the rebooting is still ongoing.
6) I have not seen Issue 1 yet with the blank or gray screen in this setup

[LogConfFiles.zip](https://github.com/ccrisan/motioneyeos/files/67889/LogConfFiles.zip)

ccrisan, 2015-12-21:
The reason why your system is rebooting is that the motion daemon is found to be hung and cannot be killed within 7 seconds (not even with the KILL signal, which indicates a kernel driver problem or a hardware issue with the device).

The "Device or resource busy" message that shows up every now and then in your motion log also indicates a problem when communicating with the webcam device. I wouldn't know what to do about that other than suggest trying with only one USB camera at a time or trying different models.

digitizing, 2015-12-22:
Well thanks for looking over my configuration and log files.  I will continue to troubleshoot further and report back my findings.   The Microsoft webcam HD-3000 I picked was supposed to be compatible with the Raspberry Wheezy OS.  I must have missed something here.

ccrisan, 2015-12-22:
It most likely is compatible. However it's not the first time I see problems when using more than one USB webcam plugged in to a device; especially when they are the exact same model. I'm no kernel/driver expert so I really wouldn't know what to do here else than try various scenarios (of camera combinations) and see what scenarios triggers the problem.

digitizing, 2015-12-22:
I tried two cameras connected locally and 3 remote network cameras (those are using just motion software) and it is rock solid for about 10 hours.  I will keep trying various combinations.

ccrisan, 2015-12-22:
Please do and keep me posted.

digitizing, 2015-12-23:
So here are some interesting observations which might help others down the road:

1) If you install on a Raspberry pi 2 two webcams with resolution 320x240 it is rock solid. (with or without powered USB hub).   As soon as you increase the resolution beyond 320x240 things get flakey.
2)If you add a third webcam to a single Raspberry pi at the 320x240 resolution things get flakey 
3)If you lower the resolution to 176x144 then I was able to get four webcams working on a single Raspberry pi.

These above tests produced similar results if you added the cameras to the MotionEye Raspberry or if you had the standard Raspian Wheezy os with the Motion software v3.2.12 added. By the way I used the Microsoft HD-3000 cameras for all of my testing.

It appears that the issue I was running into was caused by either the Raspberry pi 2 not having the horsepower or limitations with the Motion software.  So in my opinion your Motioneye software is rock solid and is not the choking point.

One minor observation.  If I added a network camera and then turned Motion Detection off (the default setting is on) and then later tried to re-enable the feature it would not work as far as sensing motion.  The fix that I used was to delete the webcam and re-install it and all was good.

Thanks for your help Calin.

ccrisan, 2015-12-24:
I'll try to reproduce the motion detection issue you mentioned. I'll keep the issue open in the meantime.

ccrisan, 2015-12-25:
I experimented a bit with netcams and motion detection and what happens in fact is that motion detection is activated with a short delay (of up to 10 seconds) for netcams, at startup. The old stable motion doesn't seem to exhibit this behaviour. Maybe a separate issue should be opened on this problem for [motion](https://github.com/Mr-Dave/motion). In any case, this is not a motionEyeOS issue (nor a motionEye one), so I'll close it for now.

closed 2015-12-25 by ccrisan